dresden-elektronik/deconz-rest-plugin #413: Rule triggering logic has a bug for operator "in"

djwlindenaar, 2018-02-12:
Hi guys,
I was wondering this morning why my lights were still on, although I put a rule in to turn them off between T01:00:00/T18:30:00. I looked into the rule triggering logic and I believe there's an issue in the code.

The offending rule:
{
    "actions": [
        {
            "address": "/groups/2/action",
            "body": {
                "on": false
            },
            "method": "PUT"
        }
    ],
    "conditions": [
        {
            "address": "/config/localtime",
            "operator": "in",
            "value": "T01:05:00/T18:25:00"
        }
    ],
    "created": "2018-02-11T10:03:47",
    "etag": "ce0f498ef1c948a6e66ef2974d6ba838",
    "lasttriggered": "2018-02-11T11:03:47",
    "name": "Night Off",
    "periodic": 0,
    "status": "enabled",
    "timestriggered": 1
}

rest_rules.cpp @1348:
        else if (c->op() == RuleCondition::OpIn && c->suffix() == RConfigLocalTime)
        {
            QTime t = now.time();
            QTime rt;

            if (rule.lastTriggered().isValid())
            {
                rt = rule.lastTriggered().time();
            }

            if (c->time0() < c->time1() && // 8:00 - 16:00
                (t >= c->time0() && t <= c->time1()))
            {
                if (rt.isValid() && rt >= c->time0() && rt <= c->time1())
                {
                    if (eItem->descriptor().suffix == RConfigLocalTime)
                    {
                        return false;  // already handled
                    }
                }
}

So the rule triggered as I created it yesterday. The rule triggering code checks lasttriggered to see if it has already triggered in the targeted timeframe, but it neglects to take the date into account. So last night at 01:05:00 it checked the lasttrigered time, found 11:03:47 and concluded that this time is within time0 and time1 timeframe. Therefore it returned false //already handled... 

I guess to solve this, we should be comparing QDateTime items, by first reconstructing the datetime from time0 and time1 and then do a check for in. 

Does that make sense? 

Best regards
Daniel

ebaauw, 2018-02-12:
Hm, there was also an issue with changed schedules not firing at the new scheduled time, when they has already fired before, see #67.

djwlindenaar, 2018-02-13:
I didn't look into schedule handling, so I don't know if that's related.

ebaauw, 2018-05-11:
I can confirm that a rule with (only) an `in` condition on `config.localtime` only fires once after being created.  This is a bug in deCONZ - I will try to issue a PR with a fix.

Also, the rule fires when deCONZ is (re-) started during the time specified, or when the rule is created during the time specified.  That's not consistent with the behaviour of the Hue bridge.  Personally, I actually prefer the deCONZ behaviour.  @manup, what's your view on this?

I also noticed that `lasttriggered` is in local time instead of UTC (as a sensor's `state.lastupdated`).  Again, that's not consistent with the Hue bridge.   I think this is a bug as well.  BTW `created` is in UTC alright. 

Also, `lasttriggered` and `timestriggered` are preserved across deCONZ restarts.  Again, that's not consistent with the Hue bridge.  Personally, I prefer the Hue bridge behaviour.  @manup, what's your view on this?

**Test case**
Two CLIPGenericFlag sensors, _Boot Time_ and _Night_ with a couple of rules:
1. Set _Boot Time_ at startup (and turn off all lights, assuming a power restore);
2. Clear _Night_ at sunrise;
3. Set _Night_ on at 23.30;
4. Clear _Night_ at 07.00.

Rules after creation on the Hue bridge [note: none triggered]:
```json
{
  "1": {
    "name": "Boot Time",
    "owner": "x",
    "created": "2018-05-11T08:12:20",
    "lasttriggered": "none",
    "timestriggered": 0,
    "status": "enabled",
    "recycle": false,
    "conditions": [
      {
        "address": "/sensors/1/state/lastupdated",
        "operator": "dx"
      },
      {
        "address": "/sensors/2/state/flag",
        "operator": "eq",
        "value": "false"
      }
    ],
    "actions": [
      {
        "address": "/sensors/2/state",
        "method": "PUT",
        "body": {
          "flag": true
        }
      },
      {
        "address": "/groups/0/action",
        "method": "PUT",
        "body": {
          "on": false
        }
      }
    ]
  },
  "2": {
    "name": "Daylight On",
    "owner": "x",
    "created": "2018-05-11T08:12:26",
    "lasttriggered": "none",
    "timestriggered": 0,
    "status": "enabled",
    "recycle": false,
    "conditions": [
      {
        "address": "/sensors/1/state/daylight",
        "operator": "eq",
        "value": "true"
      }
    ],
    "actions": [
[... 512 lines cut ...]
        "address": "/sensors/3/state",
        "body": {
          "flag": false
        },
        "method": "PUT"
      }
    ],
    "conditions": [
      {
        "address": "/config/localtime",
        "operator": "in",
        "value": "T07:00:00/T23:30:00"
      }
    ],
    "created": "2018-05-11T08:13:28",
    "etag": "6c7bdbdeb2c30ab74a7e8ac2f04cd755",
    "lasttriggered": "2018-05-11T10:22:44",
    "name": "Night Off",
    "owner": "xx",
    "periodic": 0,
    "status": "enabled",
    "timestriggered": 3
  }
}
```

ebaauw, 2018-05-12:
> > Also, the rule fires when deCONZ is (re-) started during the time specified, or when the rule is created during the time specified. That's not consistent with the behaviour of the Hue bridge. Personally, I actually prefer the deCONZ behaviour. @manup, what's your view on this?

> Hard to tell, but I figure in real world usage the current implementation has it's benefits. This might be a case there for hue and deCONZ behaviours exist use cases.

With my latest commit, actually comparing times instead of looking at `lasttriggered`, an `in` condition on `config.localtime` only triggers at start time (and `not in` only at end time).

> It should be ok to change it to UTC, but wee need to check that things which might depend on it, like ddx still work.

Changed `ddx` as well to look at actual time, instead of at `lasttriggered`.

> What about changing the timestriggered to work like hue one by init it to 0 on deCONZ re-start, but still write the current count to database for debugging purposes?

I didn't change writing `timestriggered` to the database, only disabled restoring it.  Of course, the database value will be overwritten next time the rules are saved.

Krocko, 2018-05-16:
With deCONZ 2.05.26 the following rule stop working. 
```
{
    "actions": [
      {
        "address": "/sensors/35/state",
        "body": {
          "flag": true
        },
        "method": "PUT"
      },
      {
        "address": "/groups/0/action",
        "body": {
          "on": false
        },
        "method": "PUT"
      }
    ],
    "conditions": [
      {
        "address": "/config/localtime",
        "operator": "in",
        "value": "T00:00:00/T23:59:59"
      },
      {
        "address": "/sensors/35/state/flag",
        "operator": "eq",
        "value": "false"
      }
    ],
    "name": "Boot Time",
    "periodic": 0,
    "status": "enabled"
  }
```
This rule should power off all lights, if i restart deCONZ or the Raspberry. 
@ebaauw told me to use a condition based on the daylight sensor instead of the localtime. So a change the condition. But the following rule don‘t work also. 
```
{
    "actions": [
      {
        "address": "/sensors/35/state",
        "body": {
          "flag": true
        },
        "method": "PUT"
      },
      {
        "address": "/groups/0/action",
        "body": {
          "on": false
        },
        "method": "PUT"
      }
    ],
    "conditions": [
      {
        "address": "/sensors/14/state/lastupdated",
        "operator": "dx"
      },
      {
        "address": "/sensors/35/state/flag",
        "operator": "eq",
        "value": "false"
      }
    ],
    "name": "Boot Time",
    "periodic": 0,
    "status": "enabled"
  }
```
If i am going back to deCONZ 2.05.25, the first rule is working again, but the second rule don‘t work also.

Krocko, 2018-05-16:
The following rule works with deCONZ 2.05.26

```
{
    "actions": [
      {
        "address": "/sensors/35/state",
        "body": {
          "flag": true
        },
        "method": "PUT"
      },
      {
        "address": "/groups/0/action",
        "body": {
          "on": false
        },
        "method": "PUT"
      }
    ],
    "conditions": [
      {
        "address": "/sensors/14/state/lastupdated",
        "operator": "ddx",
        "value": "PT00:00:02"
      },
      {
        "address": "/sensors/35/state/flag",
        "operator": "eq",
        "value": "false"
      }
    ],
    "name": "Boot Time",
    "periodic": 0,
    "status": "enabled"
  }
```

closed 2019-01-27 by djwlindenaar