dresden-elektronik/deconz-rest-plugin #414: Enhancement: report last scene recalled in light state.

ebaauw, 2018-02-12:
Just a note, so I won't forget about it.  Interesting request on the Hue developer's forum: report which scene is active for a light.  This information is available through attributes _Current Scene_ (0x0001), _Current Group_ (0x0002), and _Scene valid_ (0x0003) in the _Scenes_ cluster (0x0005).  Not sure I would be happy about the performance implications, issuing yet another _Read Attributes_ command while polling the lights.

While we're at it: create a Hue-API compatible `/scenes` resource, including mapping `/groups/0` to a real ZigBee group containing all lights.  This would also enable selectively restoring a scene to another group, i.e. put /groups/1/action {"scene": "xxx"} would do a group cast to group 1 to recall scene xxx of group 0.  This might actually work well with the current fast poll of member lights after a group command.

martingruening, 2018-02-20:
I would like to add one vote for this and extend this. It would be super helpful if it would be possible to query the currently active scene for a group through the API.

I use overlapping groups (A contains lights 1,2,3 - B contains light 4 - C containts lights 3,4). When I recall a scene on C in my home automation, I would love to be able to query the active scenes for A and B before, save them to a variable in the home automation and when I turn C off restore the scenes on A and B afterwards. Currently there is - to my knowledge - no way to query the active scene on a group through the API?

martingruening, 2019-07-22:
> Currently there is - to my knowledge - no way to query the active scene on a group through the API?

This looks improved. When querying a group I can find meanwhile the attribute "action/scene" that contains the last scene called on the group. This works for my use case.

ebaauw, 2019-07-22:
That's the last scene recalled through the API.  There's no guarantee that it's still active.

closed 2019-11-26 by stale[bot]

miani, 2020-07-20:
Would be super nice to get the currently active scene per group via API, including an indication that no scene is active.
Is there any solution or workaround for this?