dresden-elektronik/deconz-rest-plugin #415: Group with two white bulbs showing as XY-colormode

tleegaard, 2018-02-12:
I have a minor problem with White E27 Hue Bulbs showing up as color bulbs when grouped together.
From the API I get this..

**Group**:
```json
"3": {
"action": {
"bri": 153,
"colormode": "xy",
"ct": 0,
"effect": "none",
"hue": 0,
"on": true,
"sat": 128,
"xy": [
0.424991,
0.504986
]
},
"class": "Other",
"devicemembership": [],
"etag": "",
"hidden": false,
"id": "3",
"lights": [
"4",
"5"
],
"lightsequence": [],
"multideviceids": [],
"name": "Køkkenbord",
"scenes": [],
"state": {
"all_on": true,
"any_on": true
},
"type": "LightGroup"
}
```

**Lights**:
```json
"4": {
"etag": "",
"hascolor": false,
"manufacturername": "Philips",
"modelid": "LWB010",
"name": "Køkken V",
"state": {
"alert": "none",
"bri": 74,
"on": true,
"reachable": true
},
"swversion": "1.29.0_r21169",
"type": "Dimmable light",
"uniqueid": ""
},
"5": {
"etag": "",
"hascolor": false,
"manufacturername": "Philips",
"modelid": "LWB010",
"name": "Køkken H",
"state": {
"alert": "none",
"bri": 74,
"on": true,
"reachable": true
},
"swversion": "1.29.0_r21169",
"type": "Dimmable light",
"uniqueid": ""
}
```

tleegaard, 2018-02-13:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/342 seems to be the same issue.

roflmao, 2018-02-20:
My IKEA bulbs does the same in 2.05.05:
```
21:35:47 [rolf:~] $ curl -s 192.168.66.34/api/x/lights | jq .
{
  "1": {
    "etag": "fd43772f2536e66de1e0ba547f468f96",
    "hascolor": false,
    "manufacturername": "IKEA of Sweden",
    "modelid": "TRADFRI bulb GU10 W 400lm",
    "name": "1etg_inngang_utelys1",
    "state": {
      "alert": "none",
      "bri": 254,
      "on": false,
      "reachable": true
    },
    "swversion": "1.2.214",
    "type": "Dimmable light",
    "uniqueid": "00:0b:57:ff:fe:b1:61:ae-01"
  },
  "2": {
    "etag": "939a8444728fea2088a276f4f0f35b91",
    "hascolor": false,
    "manufacturername": "IKEA of Sweden",
    "modelid": "TRADFRI bulb GU10 W 400lm",
    "name": "1etg_inngang_utelys2",
    "state": {
      "alert": "none",
      "bri": 254,
      "on": false,
      "reachable": true
    },
    "swversion": "1.2.214",
    "type": "Dimmable light",
    "uniqueid": "00:0b:57:ff:fe:b1:3b:fb-01"
  }
}
21:35:56 [rolf:~] $ curl -s 192.168.66.34/api/x/groups | jq .
{
  "2": {
    "action": {
      "bri": 127,
      "colormode": "hs",
      "ct": 0,
      "effect": "none",
      "hue": 0,
      "on": false,
      "sat": 127,
      "xy": [
        0,
        0
      ]
    },
    "class": "Other",
    "devicemembership": [],
    "etag": "939a8444728fea2088a276f4f0f35b91",
    "hidden": false,
    "id": "2",
    "lights": [
      "2",
      "1"
    ],
    "lightsequence": [],
    "multideviceids": [],
    "name": "1etg_inngang_utelys",
    "scenes": [],
    "state": {
      "all_on": false,
      "any_on": false
    },
    "type": "LightGroup"
  }
}
```

roflmao, 2018-02-20:
My IKEA bulbs does expose xy, colormode hs(hue/sat) in group, but not in lights (These are not color bulbs) in 2.05.05:
```
21:35:47 [rolf:~] $ curl -s 192.168.66.34/api/x/lights | jq .
{
  "1": {
    "etag": "fd43772f2536e66de1e0ba547f468f96",
    "hascolor": false,
    "manufacturername": "IKEA of Sweden",
    "modelid": "TRADFRI bulb GU10 W 400lm",
    "name": "1etg_inngang_utelys1",
    "state": {
      "alert": "none",
      "bri": 254,
      "on": false,
      "reachable": true
    },
    "swversion": "1.2.214",
    "type": "Dimmable light",
    "uniqueid": "00:0b:57:ff:fe:b1:61:ae-01"
  },
  "2": {
    "etag": "939a8444728fea2088a276f4f0f35b91",
    "hascolor": false,
    "manufacturername": "IKEA of Sweden",
    "modelid": "TRADFRI bulb GU10 W 400lm",
    "name": "1etg_inngang_utelys2",
    "state": {
      "alert": "none",
      "bri": 254,
      "on": false,
      "reachable": true
    },
    "swversion": "1.2.214",
    "type": "Dimmable light",
    "uniqueid": "00:0b:57:ff:fe:b1:3b:fb-01"
  }
}
21:35:56 [rolf:~] $ curl -s 192.168.66.34/api/x/groups | jq .
{
  "2": {
    "action": {
      "bri": 127,
      "colormode": "hs",
      "ct": 0,
      "effect": "none",
      "hue": 0,
      "on": false,
      "sat": 127,
      "xy": [
        0,
        0
      ]
    },
    "class": "Other",
    "devicemembership": [],
    "etag": "939a8444728fea2088a276f4f0f35b91",
    "hidden": false,
    "id": "2",
    "lights": [
      "2",
      "1"
    ],
    "lightsequence": [],
    "multideviceids": [],
    "name": "1etg_inngang_utelys",
    "scenes": [],
    "state": {
      "all_on": false,
      "any_on": false
    },
    "type": "LightGroup"
  }
}
```

manup, 2018-02-20:
Ok as I understand it, the `group.action` should on include the superset of all `light.state` attributes?

roflmao, 2018-02-20:
Yes, the more the same the better i guess. If a group has a set of different bulbs the "weakest" link wins and sets the group settings imho.  Ex. if a colorbulb in a group with nocolorbulb then you cant set the color for a group..

tleegaard, 2018-03-12:
@manup I think including the superset is the way to go. Furthermore HA seems to take the same approach: https://home-assistant.io/components/light.group/

_"The supported features of all lights will be added together. For example, if you have one RGB light in a group of otherwise brightness-only lights, the light group will be shown with a color picker."_

tleegaard, 2018-11-23:
I'm on 2.05.48 and 262F0500. With two Philips LWB010 (white only bulbs) I get colormode hs:
```json
{
  "action": {
    "bri": 127,
    "colormode": "hs",
    "ct": 0,
    "effect": "none",
    "hue": 0,
    "on": false,
    "sat": 127,
    "scene": null,
    "xy": [
      0,
      0
    ]
  },
  "class": "Other",
  "devicemembership": [
    
  ],
  "etag": "fa732ebd57ce4da7c085a241b9167667",
  "id": "8",
  "lights": [
    "7",
    "1"
  ],
  "name": "Spisebord",
  "scenes": [
    
  ],
  "state": {
    "all_on": false,
    "any_on": false
  },
  "type": "LightGroup"
}
```
Bulb 1:
```json
{
  "etag": "10d79405ea2f333ac6280c1ffe1e8435",
  "hascolor": false,
  "manufacturername": "Philips",
  "modelid": "LWB010",
  "name": "PH5_1",
  "state": {
    "alert": "none",
    "bri": 204,
    "on": true,
    "reachable": true
  },
  "swversion": "1.29.0_r21169",
  "type": "Dimmable light",
  "uniqueid": "00:17:88:01:02:b1:b0:12-0b"
}
```
Bulb 2:
```json
{
  "etag": "10d79405ea2f333ac6280c1ffe1e8435",
  "hascolor": false,
  "manufacturername": "Philips",
  "modelid": "LWB010",
  "name": "PH5_2",
  "state": {
    "alert": "none",
    "bri": 204,
    "on": true,
    "reachable": true
  },
  "swversion": "1.29.0_r21169",
  "type": "Dimmable light",
  "uniqueid": "00:17:88:01:02:b1:40:fe-0b"
}
```
(This makes Home Assistant show the color picker, though it's not working)

tleegaard, 2019-03-25:
Issue still persists.

tomky, 2019-10-15:
Issue persists.

closed 2021-03-20 by close-issue-app[bot]

ebaauw, 2021-03-20:
This is still an issue

Mimiix, 2021-03-23:
Duplicate of https://github.com/dresden-elektronik/deconz-rest-plugin/issues/342 
Not re-opening.