dresden-elektronik/deconz-rest-plugin #416: Orvibo anyone?

almadsen, 2018-02-12:
Hi, does anyone know a way to a orvibo door/window sensor work with anything but their default bridge?

ebaauw, 2018-02-12:
They [advertise](http://www.orvibo.com/en/product/102.html) the sensor as ZigBee HA, so we should be able to add support for it to deCONZ.

I'm fascinated by their [ZigBee IR remote control](http://www.orvibo.com/en/product/31.html), but it doesn't look like they've published their manuals online.

manup, 2018-02-24:
At least the motion sensor already works with deCONZ guess the door/window sensor can be supported as well.

almadsen, 2018-03-13:
No way of making it happen?
I have one laying about, send to you if want it for testing.

Chrille2k, 2018-03-17:
I have a range of products from Orvibo (motion sensor, door sensor, ir remote control, smoke sensor and more) I have them working with the Orvibo smart hub but would like to run them with deCONZ but with no luck. Can I somehow help with implementing the support for these devices?

manup, 2018-05-31:
Can you provide some screenshots from the basic cluster read attributes. If we whitelist the modelid it should be easy to integrate.

almadsen, 2018-06-04:
@Chrille2k can you provide @manup with that info?

@manup, were do I get the data?

Chrille2k, 2018-06-04:
Sure @manup and @almadsen, but I think I would need some guidance how to obtain the info.

aronteh, 2018-07-02:
Can anyone confirm that Orvibo Motion Sensor is supported. Can find it in Deconz supported device.
Having trouble paring it as Sensors/Other in Phoscon.

manup, 2018-07-03:
>Sure @manup and @almadsen, but I think I would need some guidance how to obtain the info.

The info can be seen in the cluster info panel

[image]

>Can anyone confirm that Orvibo Motion Sensor is supported. Can find it in Deconz supported device.
Having trouble paring it as Sensors/Other in Phoscon.

I've testet the one with the modelid OJB-IR715-Z, this one works, does yours has the same modelid?

aronteh, 2018-07-03:
The model I have is ORVIBO SN10ZW Motion Sensor. I get very little info from my Cluster Info Panel.
Anything I can help to add support for this motion sensor.
[image]

This is the screen shot of the unit.
[image]

Chrille2k, 2018-07-03:
I have a Motion senor model SN11 which has the following values:

[image]

aronteh, 2018-07-03:
I am getting very inconsistent result. The basic cluster of my xiaomi motion sensor also show empty manufacturer in a few  nodes. 
[image]

But REST API dump of all the xiaomi sensor show correct manufacturer.
[image]

aronteh, 2018-07-07:
I have pair the Orvibo Motion Sensor again. Got a more complete node info, basic cluster and IAS Zone cluster.
Node Info
[image]
Basic Cluster
[image]
End Point
[image]
IAS Zone Cluster
[image]

manup, 2018-07-08:
The model ids look pretty random, that's strange never seen that before (or is a chinese character set/unicode?). The manufacturer names of the above are also different. I think in this case we need to detect the sensor based on mac address prefix and the IAS Zone cluster.

aronteh, 2018-07-09:
Yes the manufacturer name is in chinese character set.

bparcels, 2018-10-01:
I'm far too new to Deconz to know if this will solve things or not, but the gentleman [here](https://faire-ca-soi-meme.fr/test/2017/10/31/test-orvibo-smarthome-compatible-zigate/) noted that the Orvibo sensors seem fully standard - apart from the fact that they assumed the coordinator endpoint address was 0x0A instead of the standard 0x01. For his ZiGate, he basically just aliased the two addresses for the coordinator, and everything worked. Is this something achievable in deCONZ?

bparcels, 2018-10-01:
Also FWIW, Orvibo seems to whitelabel their products to at least a few other brands... In Canada, [primecables](https://www.primecables.ca/c-17889-smart-home) sell both the Orvibo, and their (at least visually) identical "LivingWise" equivalents... But they also make reference to a "LivingSmart" brand that they say will inter-operate as well. 

I've just ordered a few of the LivingWise components, so I should know more in a few days... I'm definitely curious to see what manufacturer name shows up in deCONZ.

manup, 2018-10-02:
The motion sensor looks very much like the Trust motion sensor (ZPIR-8000) which is supported since deCONZ version 2.05.39. Maybe they can share the same implementation.

bparcels, 2018-10-11:
Yep - the "LivingWise" motion sensor showed up as Orvibo, even the same model identifier as Chrille2k if memory serves (the conbee is in a pi at the moment, and I wasn't smart enough to screenshot deCONZ when first playing)... 

It doesn't show up in the web apps or anything, but does appear as a vibration sensor in the restAPI... This works fine when using the websocket->NodeRed->MQTT method... EXCEPT for the fact that after a while, the motion sensor starts to be reported as another device (i.e. when I walk past the motion sensor, I get a change event from a door (contact) sensor ID, saying "Open" (i.e. wrong ID, and wrong state altogether)?!?! Pulling the batteries from the motion sensor, and replacing, brings it back to the correct ID for a bit... but it eventually jumps back to impersonating some other sensor...

I'm not sure if I've just missed a step (again - really new to this, and haven't fully dug in yet) - perhaps I need to be enrolling the motion sensor in a zone or something first from deCONZ GUI?

At any rate - for anyone interested, the "LivingWise" (aka Orvibo) contact sensors DO work nicely  - though I believe they showed up as "Trust" or something. If the Trust motion sensor looks the same - perhaps this is yet another alias of the same actual manufacturer.

bparcels, 2018-10-11:
Sorry - minor clarification: The motion sensor showed Orvibo when looking at the deCONZ GUI... From REST, all sensors (motion and contact), show as "manufacturername": "Trust".

bparcels, 2018-10-11:
Sorry - one more bonehead error - mine are "LivingSmart", not "LivingWise"... apparently I'm doing neither!

closed 2019-02-15 by stale[bot]

liorm0505, 2019-08-07:
hi everyone, 
was there any progress with the orvibo motion sensor? 
i have a couple of these any i would like to use them with deconz.