dresden-elektronik/deconz-rest-plugin #417: Problem with ConBee Firmware Update

Nastras, 2018-02-15:
Hello,
i have the following firmware update issue with my ConBee stick.

It will be a continuous firmware update in the GUI offers, as soon as I make the firmware update, the stick is no longer reachable in deconz.

There is no way to reconnect it with deconz, only when I backup my SD card is the stick reconnected.

I also tried over the terminal to make an update and get this answer:

`pi@HomeKitServer:/usr/bin $ sudo GCFFlasher_internal -l
 ftdi_sio
GCFFlasher V2_11 (c) dresden elektronik ingenieurtechnik gmbh 2017/12/10

       1  FTDI device found

 device | vendor | product | serial  | description
--------|--------|---------|---------|----------------------
failed to open FTDI device (0) status = 3

pi@HomeKitServer:/usr/bin $ sudo GCFFlasher_internal -f
 ftdi_sio
GCFFlasher V2_11 (c) dresden elektronik ingenieurtechnik gmbh 2017/12/10
usage: GCFFlasher <options>
options:
 -r              force device reset without programming
 -f <firmware>   flash firmware file
 -d <device>     device to use
 -sn <serial>    FTDI device with serial number to use
 -l              list devices
 -h -?           print this help
done`

What can I do to fix the problem?

manup, 2018-02-15:
Here are a few things to check?

1) Which version of deCONZ are you running?
2) How do you start deCONZ &mdash; start menu, systemd deconz-gui script or commandline?
3) Please provide the output of: 

```sh
    $ ls -l /dev/ttyUSB0

    # ... and

    $ groups
```

Nastras, 2018-02-15:
Thank you for your prompt reply,

here is the data you wanted.

[image]
[image]

I hope that's enough for a first analysis?

Da ich in einem anderen Post gesehen habe das du Deutsch sprichst, können wir gerne auch Deutsch schreiben wenn du möchtest?

Gruß Nastra

Nastras, 2018-02-15:
Start is with autostart is it the gui script?

Here is a picture of the update we offered.
[image]

manup, 2018-02-16:
Lets keep it english since the audience here is mostly international :)

I'm a bit confused about the screenshots and operating system, is this a Raspbian ARM or Debian x86?

Nastras, 2018-02-16:
Ok 👍

Its Debian x86

manup, 2018-02-16:
Debian x86 is not officialy supported, albeit similar to Ubuntu there might be problems.

I've just tried to update ConBee from 0x26190500 to 0x261e0500 on Ubuntu directly from deCONZ and it worked. However the start script has some Ubuntu specific parts like check dependencies and setup the user to the dialout group, which we can change to *Debian like* parts to support both, but in your case this doesn't seem to be the problem. 

To Update manually:

    ... close deCONZ

    `$ sudo GCFFlasher_internal -d 0 -f /usr/share/deCONZ/firmware/deCONZ_Rpi_0x261e0500.bin.GCF`

Nastras, 2018-02-16:
Hello manup,

I was not aware that the debain desktop is not officially supported but it works fine now. Thanks for the hint!

I noticed two things that are probably a bug?

After a reboot the ConBee Stick does not connect by itself. Only after I removed the hardware by hand from the computer and added again at the USB port.
In addition, the connect button works in deconz although the ConBee is displayed.

Unfortunately, I still have a problem creating a user on the Homebridge-hue plugin.

I have described the problem here:

https://github.com/ebaauw/homebridge-hue/issues/289

I asked Erik, but maybe you have an idea too?

closed 2018-02-16 by Nastras