dresden-elektronik/deconz-rest-plugin #418: Osram Smart+ GU5.3 supported?

FearlessSpiff, 2018-02-15:
Hi.

I am new to all this but thought that the mentioned bulb [1] should be supported. It says it uses ZigBee LL 2.0 protocol in the documentation. What am I missing? 
I was able to pair my Ikea lights. But the Osram one doesn't show itself even after a reset. Not in the "Touchklink reset" Scan nor when I "Open Network".

Thanks!

[1] https://www.osram-lamps.com/ecatalog/smart-home/smart-home-lamps/led-reflector-lamps/smart-spot-gu5-3-tw/index.jsp

manup, 2018-02-15:
You may try the OSRAM specific reset sequence wile the network is open.

**Note** There are two sequences not sure which one works for the GU5.3.

## 1)
[image]

## 2)

[image]

FearlessSpiff, 2018-02-15:
Wow! Great! That worked. I read somewhere else to reset it with 5x ON for 3s and short OFF. Now I can control it. Thanks a lot!

closed 2018-02-15 by FearlessSpiff