dresden-elektronik/deconz-rest-plugin #419: Updating Peanut Plug firmware with deCONZ

apmon, 2018-02-16:
I have a Securifi peanut plug (https://www.securifi.com/securifi-peanut-plug) which correctly pairs with the ConBee in deCONZ. It can correctly be switched on and off, but its power reading does not seem to work at the moment. deCONZ rest plug-in does report a power value, but it is always 0

( {"etag":"222745eb58058409bd946fe3576a6610","hascolor":false,"manufacturername":"Securifi Ltd.","modelid":"","name":"Light 3","state":{"alert":"none","on":true,"power":0,"reachable":true},"swversion":"00000000","type":"On/Off light","uniqueid":"00:0d:6f:00:0a:76:e6:9d-01"}) 

According to the information I found on the SmartThings forum, it requires a firmware update.

Do you know what it would take to update the firmware with deCONZ and is the firmware available anywhere?

manup, 2018-02-17:
@ebaauw the code was almost right but stopped too early in the for loop at the first iteration :)

My OSRAM Smart+ plug now shows `state.power = 28000`.

ebaauw, 2018-02-17:
Cool!  It reports 28000 when no power is drawn, for reasons beyond my comprehension.  I'll try and fix that (checking state.on == false).

Also, we need to scale the power, either based on the divisor/mulitplier attributes or (in case of OSRAM) by whitelisting the plug.

manup, 2018-02-17:
>Also, we need to scale the power, either based on the divisor/mulitplier attributes or (in case of OSRAM) by whitelisting the plug.

I like the whitelisting approach more, keeps the REST API simple.

ebaauw, 2018-02-18:
@apmon, power should be working in the next release.  Could you please check if the plug reports power in Watts or in 0.1 Watts?  Could you post some screenshots of the deCONZ GUI with the endpoints and clusters, and the cluster info panel for the Basic and Electrical Management clusters?

apmon, 2018-02-19:
From what I have read on other forums, it sounds like this plug requires a firmware update for the measurements to work. I.e. even with the manufacturers own hub (Almond router), the plug only started reporting the power information after a firmware update and before always just reported 0. These devices, even newer ones, sound like they get shipped with the old/original/broken firmware.

I'll try and post on the manufacturers support forum to see if there is a firmware file to be downloaded somewhere.

But here are the screenshots from deCONZ:

Basic cluster:
[image]

Power configuration cluster:
[image]

Electrical Management cluster:
[image]

OTAU cluster:
[image]

ebaauw, 2018-02-19:
No _Model Identifier_?!  Not sure deCONZ can deal with that.  Could you also please post the _Node Info_ panel?  We need the _Manufacturer Code_ from there.

apmon, 2018-02-19:
Manufacturer Code is 0x1002
[image]

kzaoaai, 2019-12-24:
> I was able to update the firmware now (Had to buy Securifi's router to do that) and I am now seeing values for RMSVoltage, RMS Current and Active Power for the Peanut Plug.
> 
> The model identifier is still greyed out and empty though.
> 
> [image]
> 
> I don't know for sure what kind of load is on the plug (a laptop charger plugged in), but the Active Power * AC Power Multiplier / AC Power Divisor = 61W in the case of the screen shot, which seems roughly right. Same for the Voltage (123V sounds about right for America) and current (0.53A roughly equal to 61W / 123V)
> 
> Any chance of getting this supported by the deConz web api?

were you able to report the power measurements since updating the firmware?

shadowjig, 2019-12-28:
I got the firmware update file from Securifi for the peanut plugs.  I tried the OTA update plugin and it didn't work.  If anyone has another suggestion as to how to push the update to the devices, I'd be happy to try.  I have a SmartThings hub, Hubitat hub, and a Raspberry Pi with a raspbee module that I can use to push the update.  But both hubs are likely off limits cause the OTA features are not available to us users.

dmulcahey, 2020-07-30:
> I got the firmware update file from Securifi for the peanut plugs. I tried the OTA update plugin and it didn't work. If anyone has another suggestion as to how to push the update to the devices, I'd [...]

Do you still have this file?

shadowjig, 2020-07-31:
@dmulcahey 

Here's the file they sent me.

[Securifi_peanut_plug_firmware.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5007122/Securifi_peanut_plug_firmware.zip)