dresden-elektronik/deconz-rest-plugin #421: Visonic Sensors

Devoir666, 2018-02-17:
I'm new to this forum sorry if I'm asking in the wrong location.

Can someone add the Zigbee   sensors manufactured by Visonic     http://catalog.visonic.com/   ?   

Next K85 SMA   motion sensor   

http://www.visonic.com/Data/Uploads/Next_PG2_Next_K9_PG2_English_Installer_Guide_D-302413.pdf

MCT-320 SMA   door sensor

Many thanks 

Devoir

manup, 2018-02-18:
According to the datasheet and online catalog the K9-85 isn't a ZigBee sensor. Only the devices listed under ZigBee can be integrated. At least it seems to be possible since they use ZigBee HA 1.2 standard. The battery life of these is quite interesting.

Devoir666, 2018-02-18:
manup,

Thanks for the update.....

Both sensors are used on the Xfinity security system. .    I though both would have been zigbee since the MCT-320 was zigbee.  the Technicolor model TAC203COM Hub must have multiple wireless capability.  

I haven't changed the batteries in three years.

Devoir

manup, 2018-02-18:
Yes I think they use a proprietary wireless protocol for some products as well as ZigBee.

manup, 2018-02-20:
I close the issue since the device can't be integrated.

closed 2018-02-20 by manup