dresden-elektronik/deconz-rest-plugin #422: deconz auto connect error

Razeeer, 2018-02-18:
Hello,
i have installed deConz in the 2.05.02 with me everything seems to work. I noticed this thing here after using the command "htop".

[image]

Is that a significant mistake or maybe ab bug?

manup, 2018-02-18:
Nope all good, that are just the commandline parameters `--dbg-error=1` instructs deCONZ to log errors to journald service :)

Razeeer, 2018-02-18:
Hey, thanks for the quick reply.

I still have a question to my conbee stick will be recognized after a reboot only if I short disconnect from the computer.

Is there a way to change that or a terminal command restart this usb port?

manup, 2018-02-18:
Which platform are you using, Raspbian or Ubuntu? Normally no physical disconnect should be needed. Just make sure that the USB dongle is not attached to some other service or captured by a virtual machine, for example VirtualBox likes to "take" devices.

Razeeer, 2018-02-18:
I use Raspbian, how can I tell that he is not stolen by any other software,
i only have homebridge as a service and deconz with gui on the run?

manup, 2018-02-20:
I don't know if homebridge or maybe a plugin of it may also establishes a connection to USB dongle.
To gain more data you can start deCONZ manually with debug output enabled, when the problem occurs again:

1) close deCONZ
2) start with logging

    `$ deCONZ --dbg-error=1 --dbg-info=2`

Nastras, 2018-02-21:
Hello @manup,

I had you written the days that I also have the problem that the Conbee Stick does not connect automatically, so I'll take part here. In addition, I still have the problem that the startup of deCONZ not work since the update to 2.05.04 /deconz/ubuntu/beta

I was looking for the problem and found the following things:

1. When I get the status of "sudo systemctl status deconz-gui" I get this error
[image]

2. If I open manually after a reboot deCONZ it looks like this:
[image]

Here I can press the connect button but nothing changes. When I enter via the "sudo systemctl stop deconz-gui" terminal, ConBee immediately connects to deCONZ and the network is established.

3. In the Start menu the link did not work either, only after I removed the "-autostart.sh" from usr / bin / deCONZ
[image]

4. I also tested this command "deCONZ --dbg-error = 1 --dbg-info = 2" and get this answer:
[image]

I suspect that my problem is related to the connection of the ConBee and the non-working startup of deCONZ.

Do you have any idea how we can solve the problem or is it a bug?

manup, 2018-02-21:
>Here I can press the connect button but nothing changes. When I enter via the "sudo systemctl stop deconz-gui" terminal, ConBee immediately connects to deCONZ and the network is established.

It looks very much like multiple deCONZ instances running at the same time. 
`$ ps ax | grep /usr/bin/deCONZ`

The deCONZ-autostart.sh script prevents that, please don't start deCONZ directly without checking if it is already running.

I figure if you don't see the already running deCONZ after boot on desktop that you may have multiple X sessions running when logging in, this can happen for some VNC/RDP servers. Or in short deCONZ runs in another X session when the one you're logged in and looking at.

Nastras, 2018-02-21:
Thank you for your prompt reply!

Since I use Tightvnc and there an extras Session is started, I thought only that it could be really because.

I just tested it. After a reboot.
 deCONZ is not displayed. If another instance should already be started here, the conbee should be accessible via Phoscon. Unfortunately he is not.

1. Step: After Reboot
2. Step after Stop systemctl
3. Step deConz start manuell over GUI startmenu
4. Step deCONZ is running

[image]

Nastras, 2018-02-22:
Hey @manup,

I found the error, the systemd service of deCONZ-gui has been disabled.

`pi@HomeKitServer:~ $ sudo systemctl status deconz-gui
● deconz-gui.service - deCONZ: ZigBee gateway -- GUI/REST API
   Loaded: loaded (/etc/systemd/system/deconz-gui.service; disabled; vendor preset: enabled)
   Active: inactive (dead)`

I have activated this now it works now.
`sudo systemctl enable deconz-gui` now it works!

In addition, the error with the VNC server was not wrong, i have now switched from Tightvnc to X11VNC there is session 0 started by the monitor

Thank you!

closed 2018-02-26 by Razeeer

chimpy, 2019-01-08:
Hey @Nastras could you please detail how you got x11vnc working with deconz-gui? I am an X11 noob. Cheers!

dape82, 2019-01-20:
Hier ist das Log. Vom Start, bis nach dem ersten Reconnect.

```
19:59:27:787 HTTP Server listen on address 0.0.0.0, port: 80, root: /usr/share/deCONZ/webapp/
19:59:27:787 CTRL. 3.16.219:59:27:928 ZCLDB init file /home/pi/.local/share/dresden-elektronik/deCONZ/zcldb.txt
19:59:28:195 parent process bash
19:59:28:195 gw run mode: normal
19:59:28:196 GW sd-card image version: Phoscon-GW, V4_00, 2018-10-30
19:59:28:196 sd-card cid: 1b534d454434515430ffffffff012b49
19:59:28:198 DB sqlite version 3.16.2
19:59:28:199 DB PRAGMA page_count: 30
19:59:28:199 DB PRAGMA page_size: 4096
19:59:28:200 DB PRAGMA freelist_count: 0
19:59:28:200 DB file size 122880 bytes, free pages 0
19:59:28:200 DB PRAGMA user_version: 6
19:59:28:200 DB cleanup
19:59:28:200 DB create temporary views
19:59:28:228 don't close database yet, keep open for 900 seconds
19:59:28:229 started websocket server at port 443
19:59:28:234 discovery updated announce interval to 10 minutes
19:59:28:238 found node plugin: libde_rest_plugin.so - REST API Plugin
19:59:28:243 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
19:59:28:265 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
19:59:28:555 discard sensor state push for 1: state/dark (already pushed)
19:59:28:557 discard sensor state push for 1: state/status (already pushed)
19:59:28:558 discard sensor state push for 1: state/lastupdated (already pushed)
19:59:28:930 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
19:59:28:930 device state timeout ignored in state 2
19:59:29:182 UPNP socket not bound, state: 0
19:59:29:466 could not delete /var/tmp/deconz-firmware-version
19:59:29:584 Device firmware version 0x262F0500
19:59:29:610 unlocked max nodes: 200
19:59:29:898 Device protocol version: 0x0109
19:59:29:934 new node - ext: 0x00212effff022cc7, nwk: 0x0000
19:59:30:042 SensorNode 4 set node 0x00158d0002375fe6
19:59:30:042 SensorNode 2 set node 0x00158d0002375fe6
19:59:30:043 SensorNode 3 set node 0x00158d0002375fe6
19:59:30:086 SensorNode 5 set node 0x0017880104e6e470
19:59:30:104 SensorNode 6 set node 0x00158d0002b5c578
19:59:30:105 SensorNode 8 set node 0x00158d0002b5c578
19:59:30:105 SensorNode 9 set node 0x00158d0002b5c578
19:59:30:126 SensorNode 10 set node 0x00158d0002b7eea3
19:59:30:127 SensorNode 11 set node 0x00158d0002b7eea3
19:59:30:159 SensorNode 7 set node 0x00158d0002c7ee71
19:59:30:363 Current channel 25
19:59:30:409 CTRL got nwk update id 0
19:59:30:426 CTRL ANT_CTRL 0x03
19:59:30:538 Device protocol version: 0x0109
19:59:30:729 Current channel 25
19:59:30:777 CTRL got nwk update id 0
19:59:30:794 CTRL ANT_CTRL 0x03
19:59:33:479 scan skip host .100
19:59:33:515 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_Rpi_0x262f0500.bin.GCF
19:59:33:515 GW firmware version: 0x262f0500
19:59:33:515 GW firmware version is up to date: 0x262f0500
19:59:33:718 Announced to internet
19:59:33:719 discovery server date: Sun, 20 Jan 2019 18:59:33 GMT
19:59:33:719 	 local time seems to be ok
19:59:33:719 discovery found version 2.04.35 for update channel stable
19:59:40:981 scan finished
19:59:43:658 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
19:59:43:658 device state timeout ignored in state 3
19:59:45:577 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
19:59:45:577 device state timeout ignored in state 3
19:59:46:577 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
19:59:46:577 device state timeout ignored in state 3
19:59:47:577 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
19:59:47:577 device state timeout ignored in state 3
19:59:48:577 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
19:59:48:577 device state timeout ignored in state 3
19:59:49:577 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
19:59:49:577 device state timeout (handled)
19:59:50:109 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
19:59:50:109 device state timeout ignored in state 2
19:59:50:594 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
19:59:50:594 device state timeout ignored in state 2
19:59:51:092 could not delete /var/tmp/deconz-firmware-version
19:59:51:164 Device firmware version 0x262F0500
19:59:51:209 unlocked max nodes: 200
19:59:51:492 Device protocol version: 0x0109
19:59:51:685 Current channel 25
19:59:51:732 CTRL got nwk update id 0
19:59:51:749 CTRL ANT_CTRL 0x03
19:59:51:861 Device protocol version: 0x0109
19:59:52:052 Current channel 25
19:59:52:100 CTRL got nwk update id 0
19:59:52:117 CTRL ANT_CTRL 0x03
19:59:54:269 saved node state in 1 ms
19:59:54:678 sync() in 408 ms
19:59:54:678 REST API plugin shutting down
19:59:54:778 sync() in 98 ms

```

boutermw, 2022-06-21:
> I have activated this now it works now.
> `sudo systemctl enable deconz-gui` now it works!

This worked for me! Thank you!!