dresden-elektronik/deconz-rest-plugin #423: Power consumption reporting through `state.power`

ebaauw, 2018-02-18:
v2.05.04 should support power consumption through the _Electrical Measurement_ cluster.  A new attribute, `state.power` should appear in the `/lights` resource for plugs (and lights, switches, dimmers, etc) that provide this cluster.  It should report the current power consumption in Watt.

I've got this working for the Heiman smart plug and OSRAM Smart+ (new) and LIGHTIFY (old) plugs.  The Heiman _does_ support attribute reporting on _Active Power_, but the OSRAMs don't.

Some remarks:
- It doesn't work for my ubisys dimmer, presumably because it exposes the _Electrical Measurement_ cluster on a different endpoint with device type _Metering Device_, which is not whitelisted in `addLightNode()`;
- For plugs than don't support attribute reporting on _Active Power_, deCONZ needs to poll the device to update `state.power`.  It does a forced poll, after you switch the plug on or off, but if you just change the consumption of the attached device, without changing `state.on` of the plug, it might take some time before `state.power` is updated;
- No support yet for the Xiaomi plug, which reports power through an _Analog Input_ cluster, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/404#issuecomment-364779685.  I suspect the Xiaomi in-wall switches also report power this way, see #335 and #356.

@manup, I'm not sure if we can support the ubisys, Xiaomi, and others, which provide the _Electrical Measurement_ (or _Analog Input_) cluster on a different endpoint than the _OnOff_ cluster.  Maybe we should expose power as a ZHAPower `/sensors` resource instead, leveraging the sensor whitelist and ignoring the device type.  What are your thoughts on this?

manup, 2018-02-18:
2.05.04 is online now for all platforms.

> @manup, I'm not sure if we can support the ubisys, Xiaomi, and others, which provide the Electrical Measurement (or Analog Input) cluster on a different endpoint than the OnOff cluster. Maybe we [...]

Albeit I like the simplicity of `state.power` in `/lights` as a API user, a separate ZHAPower fits better in existing way of splitting up resources into various sensors.

ebaauw, 2018-02-18:
> 2.05.04 is online now for all platforms.

Thanks!

> Albeit I like the simplicity of `state.power` in `/lights` as a API user, a separate ZHAPower fits better in existing way of splitting up resources into various sensors.

If the plug supports attribute reporting, this would probably be easier, but if it doesn't we'd need to poll the sensor.  It's mains powered, so no big deal, but no easy poll manager.

Anyways, it seems kinda academic: the ubisys reports something, but definitely not the consumption of the attached lights.  It varies from 12 (when the dimmer is off) to 5 (when the connected lights are at full birghtness).  Maybe the power drawn by the dimmer itself?  The OSRAM plugs seems to report 28000 when the power is off, and 400 when their power is on, irrespective what power the  the connected device is drawing (or even if a device is connected).  It works brilliantly for the Heiman, though.

closed 2018-03-02 by ebaauw

bdb12, 2019-04-25:
Hi Guys

So from what I can read here and with the links the Xiaomi (Lumi.plug) is still not supported in terms of consumption? I am using HA and while I see a consumption entity I cannot see any data in HA. I

Thanks in advance guys :)

bslefevre, 2019-10-22:
Hi, I'm using the OSRAM Smart+plugs but Phoscon API is returning me this: (without state.power)
` {
        "etag": "randomletters",
        "hascolor": false,
        "manufacturername": "OSRAM",
        "modelid": "Plug 01",
        "name": "On/off - Zolder",
        "state": {
            "alert": "none",
            "on": true,
            "reachable": true
        },
        "swversion": "V1.04.90",
        "type": "On/Off plug-in unit",
        "uniqueid": "xx:xx:xx:xx:xx:xx"
    }`

Is there anything I can do to get state.power shown (and ultimately visible in HASS)?