dresden-elektronik/deconz-rest-plugin #427: Samsung SmartThings Arrival Sensor

theendisnye, 2018-02-20:
Does the Samsung SmartThings Arrival Sensor work with DeConZ or is that planned in the future?

Steve

wvuyk, 2018-02-21:
For Europe the name seems to be Samsung smartthings Presence F-PRS-UK-v2 Sensor. They are quite interesting as they also perform as router?

manup, 2018-02-21:
>They are quite interesting as they also perform as router?

On a small CR2032 battery that's unlikely :) routers usually are mains powered.

theendisnye, 2018-02-21:
Here is a link to the product http://www.samsung.com/uk/smartthings/sensors-plug-f-prs-uk-v2/ I think they basically send out a regular 'I am here' poll to maintain the presence status and then when Smartthings stops getting those polls for a period it sets the device to departed setting it back to arrive when they are re-instated. I was offered really cheap,  which I now have. I have tried to include it in DeConZ which seems to see it but does not seem to do anything with it.

theendisnye, 2018-02-23:
From what I can tell this is the end point information.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

manup, 2018-02-23:
Does the sensor return anything when `read` button pressed in `Basic Cluster`?
Knowing the `Model Identifier` attribute would be helpful.

theendisnye, 2018-02-23:
It returns manufacturer SmartThings and Model Identifier tagv4

Steve

uldisbiks, 2018-03-14:
Does it also shows up in REST API? I`m trying to connect presence sensor to Home Assistant trough deCONZ and I can see sensor in Xwindow APP but REST does not return sensor info.

theendisnye, 2018-03-14:
I am using homeseer with the Jowihue plugin and that does not seem to recognise the device. I assume it is not getting anything from the API.

Steve

manup, 2018-03-29:
Hi, does the binary input attribute change when presence is detected?

cowinger, 2018-03-30:
I also use homeseer with the jowihue plugin and would like to have this arrival sensor work. I did not see any changes to the binary input when it was removed from the area.

uldisbiks, 2018-04-01:
I can send sensor to someone to make better and faster dev

wvuyk, 2018-06-09:
I have been playing here wih a smarthings arrival sensor. @manup  Indeed there is no change in binary input. I hink the whole trick here is that the sensor is reachable or unreachable? If you test the Identify function in deCONZ, the sensor is beeping - would be nice to enable this in the REST API. 
There is a challenge with the unreachable function as indication if the car is in or out of reach.... it would be nice if unreachable detection for this device would be set faster then for other devices maybe?

I guess enabling this device for REST API could be simple - although adding the identify function as locating the device (beep) might be a bit more tricky? I also would have sensor to test for one of the developers?

ebaauw, 2018-06-09:
Supporting sending _Idenfity_ to the sensor will be straightforward.   We already do this for e.g. the Hue motion sensor through `config.alert`.

I don't think I've come across the _Binary Input_ cluster before.  As _Present value_ is read-write, it might be used for configuring the sensor, rather than reporting the state. I cannot seem to find the usual SmartThings custom config, probably because it's a standard SmartThings sensor supported out of the box.  I'm afraid we need to do some sniffing.  @uldisbiks, I'll be happy to take a look at the sensor and return it afterwards, if you're willing to send it to Amsterdam.

> I think they basically send out a regular 'I am here' poll to maintain the presence status and then when Smartthings stops getting those polls for a period it sets the device to departed setting it back to arrive when they are re-instated.

I would theorise that the "I am here" is actually an _Identify_ command, given the _Identify_ client cluster (in addition to the server cluster). If that's indeed the case, we could probably leverage the buttonhandler for switches, exposing the sensor as a (stateless) switch.  You could than create a `ddx` rule on `state.lastupdated` to detect that the sensor hasn't checked in for a while, setting a corresponding CLIPPresence companion sensor to false.  And a regular switch rule to set the CLIPPresence to true once the sensor checks in again.

BTW I'm using HomeKit geofencing to set some CLIP sensors on deCONZ (that homebridge-hue exposes as switches to HomeKit).  It works most of the time, but it's rather coarse, based on the location determination of your iPhone.  A solution based on reachability of the ZigBee network might be more precise, if it indeed has the advertised range of 27m.

wvuyk, 2018-06-09:
@ebaauw I can send it to you fine. I will need the address to send it to. Could you send this to w dot vuyk at upcmail dot nl?

manup, 2018-06-09:
Looks like the sensor is even more creative than the xiaomi devices :)

ebaauw, 2018-06-13:
Or not...

Received the sensor.  It pairs to the deCONZ GUI without any issues.  Indeed, it beebs when you send the _Identify_ command.  It seems quite responsive to reading characteristics - for a battery-powered device.

[image]

It only reports battery voltage, not percentage:
[image]

Every 10 minutes (588 seconds), the sensor sends a couple of _ZDP Match Descriptor Request_ messages (protocol 0x0000, cluster 0x0006), specifying a single input cluster 25 (0x0019, the _OTAU_ cluster), and no output clusters.  The RaspBee responds with matching endpoint 1.  Then, the sensor sends five _Query Next Image Request_ messages (profile 0x0104, cluster 0x0019), to which the RaspBee responds with an ACK.  
I never sniffed the OTAU sequence before, but @manup, I take it this is the normal way for a device to check whether it's firmware can be upgraded?  As checking every 10 minutes seems ridiculously often, I assume the sensor (ab)uses this for checking in. 

deCONZ seems to catch all these messages already, updating the last seen indication:
```
Jun 13 19:34:41 pi1 deCONZ[10162]: 19:34:38:702 Node 0x24FD5B000105AB46 is known by 0 neighbors, last seen 587 s
Jun 13 19:34:42 pi1 deCONZ[10162]: 19:34:40:143 APS-DATA.indication srcAddr: 0x24fd5b000105ab46, dstAddrMode: 2, profile: 0x0000, cluster: 0x0006, lqi: 255, rssi: -33
Jun 13 19:34:42 pi1 deCONZ[10162]: 19:34:40:622 Node 0x24FD5B000105AB46 is known by 0 neighbors, last seen 0 s
Jun 13 19:34:47 pi1 deCONZ[10162]: 19:34:42:107 APS-DATA.indication srcAddr: 0x24fd5b000105ab46, dstAddrMode: 2, profile: 0x0104, cluster: 0x0019, lqi: 255, rssi: -34
Jun 13 19:34:47 pi1 deCONZ[10162]: 19:34:42:542 Node 0x24FD5B000105AB46 is known by 0 neighbors, last seen 0 s
Jun 13 19:34:47 pi1 deCONZ[10162]: 19:34:44:462 Node 0x24FD5B000105AB46 is known by 0 neighbors, last seen 2 s
Jun 13 19:34:47 pi1 deCONZ[10162]: 19:34:45:034 APS-DATA.indication srcAddr: 0x24fd5b000105ab46, dstAddrMode: 2, profile: 0x0104, cluster: 0x0019, lqi: 255, rssi: -33
Jun 13 19:34:47 pi1 deCONZ[10162]: 19:34:45:162 int zmMaster::readParameter(ZM_DataId_t) 0x24
Jun 13 19:34:47 pi1 deCONZ[10162]: 19:34:46:382 Node 0x24FD5B000105AB46 is known by 0 neighbors, last seen 1 s
Jun 13 19:34:50 pi1 deCONZ[10162]: 19:34:47:971 APS-DATA.indication srcAddr: 0x24fd5b000105ab46, dstAddrMode: 2, profile: 0x0104, cluster: 0x0019, lqi: 255, rssi: -34
Jun 13 19:34:50 pi1 deCONZ[10162]: 19:34:48:302 Node 0x24FD5B000105AB46 is known by 0 neighbors, last seen 0 s
Jun 13 19:34:50 pi1 deCONZ[10162]: 19:34:50:222 Node 0x24FD5B000105AB46 is known by 0 neighbors, last seen 2 s
Jun 13 19:34:50 pi1 deCONZ[10162]: 19:34:50:922 APS-DATA.indication srcAddr: 0x24fd5b000105ab46, dstAddrMode: 2, profile: 0x0104, cluster: 0x0019, lqi: 255, rssi: -33
Jun 13 19:34:54 pi1 deCONZ[10162]: 19:34:52:142 Node 0x24FD5B000105AB46 is known by 0 neighbors, last seen 1 s
Jun 13 19:34:54 pi1 deCONZ[10162]: 19:34:53:849 APS-DATA.indication srcAddr: 0x24fd5b000105ab46, dstAddrMode: 2, profile: 0x0104, cluster: 0x0019, lqi: 255, rssi: -34
Jun 13 19:34:54 pi1 deCONZ[10162]: 19:34:54:062 Node 0x24FD5B000105AB46 is known by 0 neighbors, last seen 0 s
```

To support the sensor in deCONZ, I guess we need a hack when the last seen indication is updated, to set `state.presence` to `true`.  Then with a `config.duration` set to 600 (10 minutes), `state.presence` should revert to `false` when the check in is missed.  Also, we need to translate the battery voltage to percentage for reporting in `config.battery`.  Haven't seen any battery reports yet, though.  Maybe I need to setup attribute reporting first...

ebaauw, 2018-06-13:
Good news, no need for a hack!  The sensor does support attribute reporting for the_Binary Input_ cluster's _Present Value_ attribute (0x000f/0x0055) and for the _Power Configuration_ cluster's _Battery Voltage_ attribute.  The _Present Value_ remains false, but deCONZ could update `state.presence`, ignoring the value.

Currently set the report interval to every five minutes - seems to work.  I suppose I should take the sensor out for a walk, and see if the reporting resumes after coming back into network range.

wvuyk, 2018-06-14:
that is hopeful news :-) It means it can take up to 5 minutes to see if you are out? Am very curious how long it takes when you come back... am hoping it could be fast becuase it reports once connects I suppose?

ebaauw, 2018-06-14:
> It means it can take up to 5 minutes to see if you are out?

Depending on the frequency set in the report configuration.  Classical tradeoff between battery life and responsiveness.  I now reports 2.4V, where it reported 2.9V when I first paired it yesterday.

> am hoping it could be fast becuase it reports once connects I suppose?

Indeed, when coming into range, the sensor issues a report immediately after the device announcement.  I'm not sure yet how quickly it detects that it's in range.  Easier to test once I've added support to deCONZ and homebridge-hue.

ebaauw, 2018-06-14:
Please check out my latest PR.  The sensor should supported:
```json
{
  "config": {
    "alert": "select",
    "battery": 40,
    "duration": 310,
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "44c7120057cef1c8abf2665833f55d9b",
  "manufacturername": "SmartThings",
  "modelid": "tagv4",
  "name": "Presence 10",
  "state": {
    "lastupdated": "2018-06-14T19:58:11",
    "presence": true
  },
  "swversion": "R743_#33",
  "type": "ZHAPresence",
  "uniqueid": "24:fd:5b:00:01:05:ab:46-01-000f"
}
```

Remarks:
- I'm not sure about `config.battery`, as the sensor only reports battery voltage.  I translate it to percentage assuming a linear scale from 3.0V (100%) to 2.0V (0%).  Max value I've seen is 2.9V and minimum 2.4V.  Please report what values you're seeing.  I setup reporting for the battery voltage once every hour, so you might need to wait a while before `config.battery` is updated.  Alternatively, you can manually read the _Power Configuration_ cluster in the deCONZ GUI;
- Attribute reporting should be setup for 0x000f/0x0055 (_Binary Input_/_Present Value_), to send a report every 300 seconds.  Only after the reporting has been setup, the sensor will report `state.presence` as true.  I initialise `config.duration` to 310 seconds, to reset `state.presence` to false when the next report hasn't been received;
- If you're feeling lucky, you might alter the reporting setup in the deCONZ GUI and adjust `config.duration` accordingly.  If you specify a longer interval, deCONZ will probably overwrite it, sooner or later, but if you specify a shorter interval, deCONZ might actually leave that be.  Not sure how badly this will effect the battery life;
- Set `config.alert` to `"select"` or `"lselect"` to make the sensor beep once or 12 times;

wvuyk, 2018-06-14:
Will be testing, but also have to wait for a build as I am still left handed on PR builing :-)

ebaauw, 2018-06-15:
First overnight test.  Eve reports motion for the last 10:25 hours, so that's looking good.

Battery is down to 2.2V.  It's a regular CR2032, which is good news: these are damn cheap at IKEA (look for PLATTBOJ).  There's also a small push button under hood - presumably to reset the sensor.

I will bring the sensor along some errands this afternoon and see how that goes.

ebaauw, 2018-06-15:
Hm, it took the sensor a good 15 minutes to report back in after coming home.  Granted, this is on my test network with only two routers besides the RaspBee (a Xiaomi in-wall switch and an innr smart plug).

wvuyk, 2018-06-18:
I will install .30 today and test it here also. But if it is taking this long it would miss the function 'arrival' really for what I want to use it. Maybe then still the reachable/unreachable should be used as arrival indicator?

Wim

wvuyk, 2018-06-18:
Ok, was hoping the rejoin would happen once a router could be found by the arrival sensor. For my plugin I do not use rules at all. The logic is done by my plugin in combination with Homeseer, so I could catch the unreachable/reachable state if needed.
Will do my own test soon here too.

wvuyk, 2018-06-18:
Tried to update my raspbee here to 2.05.30, but after updating and restyarting deCONZ does not start anymore? What changed that might prevent the autostart? Coming from 2.05.28 here

manup, 2018-06-18:
Is there something printed on the console?
Also if you compile the plugin on your own, make sure the libsqlite3-dev package is installed.

wvuyk, 2018-06-18:
also the update to 2.05.29 fails, returning to 2.05.28 works fine. Is this libsqlite3-dev new since .29?

wvuyk, 2018-06-18:
Ok, adding the package  libsqlite3-dev works, I am now getting deCONZ started. there was no errors on the console BTW and also before installing the package I did not see errors during compile, but could have missed those easily ....

wvuyk, 2018-06-18:
I have been waiting for a long time now, but no updates are coming in. I tried to do a read of the clusters and then an update came in setting presence to true, did set duration to 90 for testing and presence went to false after a bit, while the arrival sensor was still in the house. Went out for some time and came back in. No updates anymore here?

` "6": {
        "config": {
            "alert": "none",
            "battery": 60,
            "duration": 90,
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "e667f31e996d9d828c7c4281b6d8f6bc",
        "manufacturername": "SmartThings",
        "modelid": "tagv4",
        "name": "Arrival sensor",
        "state": {
            "lastupdated": "2018-06-18T10:58:51",
            "presence": false
        },
        "swversion": "R743_#33",
        "type": "ZHAPresence",
        "uniqueid": "24:fd:5b:00:01:05:ab:ae-01-000f"
    },`

ebaauw, 2018-06-18:
Did the deCONZ REST API plugin setup the attribute reporting?  I cannot quite put my finger on it, but this might take some time and/or restarts of deCONZ.  Check in the deCONZ GUI, by double clicking on the _Present Value_ attribute of the _Binary Input_ cluster in the _Node Info_ panel, and pressing _read config_.  
[image]

You might try and configure this manually, and, in the _Bind Dropbox_ panel create a binding from the cluster to the RaspBee.
[image]

wvuyk, 2018-06-19:
I have checked the reproting configuration and it was set to 300 indeed. I have reset the config.duration to 310, as this might be confusing the updates here. Also did several restarts of deCONZ and checked the arrival sensor several times. During a read the lastupdated time is set and the presence set to true, but after 310 seconds the presence in REST API is reset to false. After that no updates on the lastupdated, so I get the feeling the reporting is not done?

If I try to check the bind Dropbox I can only change IEEE to Group, but I cannot change anything here?

Wim

ebaauw, 2018-06-19:
No.  The only way to check the. Inding is to delete it - if it displays success there was a binding, otherwise you’ll get a binding not found (or similar) message.  Make sure to (re-)bind afterwards.

wvuyk, 2018-06-19:
Sorry to be so ignorant here, but how do I delete a binding? I do not see if it is bound, nor can I acces a unbind button... everything is greyed out here?

[image]

wvuyk, 2018-06-19:
Can this be caused by the fact that this sensor already was joined in the earlier version of deCONZ? Should I have deleted it from deCONZ, or maybe even from the deCONZ SQLite database?

ebaauw, 2018-06-19:
> how do I delete a binding? I do not see if it is bound, nor can I acces a unbind button... everything is greyed out here?

You need to drag the source cluster to the _Source_ dropbox and the destination end point to the _Destination_ dropbox.  Then the fields are populated as on my screenshot, and the _Bind_ and _Unbind_ buttons become active.

I don’t think the binding has been created by deCONZ.  300/300 might be the sensor’s factory default reporting setting for the _Present Value_ attribute.  DeCONZ should set it to 10/300: https://github.com/dresden-elektronik/deconz-rest-plugin/blob/a60076bd096269c8dd81aa78841c2aa1c2558720/bindings.cpp#L832-L839

wvuyk, 2018-06-21:
I did as you said and managed to setup the binding. Now the REST API is updating the status as defined. 
It could be that I have set the 300/300 setting, I tried it once but deCONZ reported an write error to me. Guessing there was no write error after all then....
For me also the report of arrival sensor came in quite some time after the real arrival of me. Although this might be a trustworthy report, it might be too late for switching the lights on arrival. 

One question left her for me is why deCONZ did not create the binding when the REST API was reporting the device? Why is this the first time I had to do a binding here? Could this have been caused by the fact that this arrival sensor was already connected to deCONZ before?

ebaauw, 2018-06-22:
> why deCONZ did not create the binding when the REST API was reporting the device?

Not sure, maybe I screwed up something, I’ll do some more testing and sniffing, but it definitely created the binding and attribute reporting for the battery for me.

deCONZ should check and setup the binding at device discovery and every so often.  In my experience, this is more reliable for mains-powered devices than for battery-powered devices.  For the latter, deCONZ will only send the request once it knows the device is awake, because it just received a message from the device.  Of course, it will only receive messages after the binding and reporting have been setup.  There’s a small window of opportunity to break this catch 22 when the device is paired or first announces it presence, but it seems to be a bit of a hit and miss.  Of course all the ZigBee I/O is asynchronous and sharing the same queues on the RaspBee or ConBee, so it’s hard to figure out what’s happening when exactly.

ebaauw, 2018-06-23:
> maybe I screwed up something

I found another place where clusters are whitelisted for attribute reporting, where the _Binary Input_ cluster was missing.  After adding it, deCONZ now sets up the reporting correctly.

wvuyk, 2018-06-23:
Will test this when available! 🥇

wvuyk, 2018-06-28:
Tested this yesterdy by unbinding the sensor in 2.05.29, installing 2.05.31 and the sensor ws boud again (reporting as expected). I gues it is working now! 
Who should close this issue now? @theendisnye ?

Wim

theendisnye, 2018-06-28:
Thanks Guys, I don't pretend to understand all the technical content but I can't wait to use it.

Steve

closed 2018-06-28 by theendisnye

holdestmade, 2019-03-21:
This is not being discovered in the Phoscon App, is there any way it can be added ?

Thanks