dresden-elektronik/deconz-rest-plugin #429: QUESTION - Time based rules for motion detection and ability to use scenes?

Martin-Gardner, 2018-02-21:
Is it possible to have different time rules for the hue motion detection and the ability to select different scenes on triggers?

Martin-Gardner, 2018-02-21:
Also, is it possible to reassign buttons on remotes like the hue dimmer and the Ikea switch?

Martin-Gardner, 2018-02-21:
Just looking at the REST API, I assume that I would be setting up rules for the triggers?

manup, 2018-02-21:
Yes one rule rule for each time range, which is specified in a rule condition.
For the switches you can define rules which trigger you custom actions on receiving button events.

Martin-Gardner, 2018-02-21:
Ok - I will give it a go - Cheers

balu-, 2018-02-21:
if anyone of you has a specific example i would very much apriciate it 
(also other usefull rule examples)

carloscae, 2019-01-02:
Hey everyone. Any news of this feature coming to Phoscon? I'm looking into mimicking the nightlight function of Hue. Since I'm running this on Hassio, no access to Rest API. Cheers.

manup, 2019-01-02:
The motion sensor configuration will get a major refactor this year. It will include way more options to specify the time range as well as selecting scenes to trigger. Unfortunately this is not around the corner. Currently rules created with the REST-API are the only option.

>Since I'm running this on Hassio, no access to Rest API. Cheers.

The REST-API should be reachable there too on the configured port? Otherwise Phoscon App couldn't communicate with deCONZ.

carloscae, 2019-01-02:
Thanks for the feedback!
 
> The REST-API should be reachable there too on the configured port? Otherwise Phoscon App couldn't communicate with deCONZ.

Hm... I think I'm stupid and confusing Rest API with the GUI. I'll look into this.

closed 2019-05-09 by stale[bot]

imbaimbaimba, 2019-07-10:
@manup :
Any news on the motion sensor refactor? 

What I would basically like to do, is just be able to activate different scenes with a motion sensor, inside different time spans. Also the built-in functionality to by-pass sensor activation when a wall switch is used, is also something I would like to include in this - but maybe a bit alternatively.

This leads me to some confusion around assigning wall switches to lights. I got IKEA Remotes that have been custom built into wall switches (only on/off, dim up, dim down and scene forward - scene backwards is left out). If I go to the light group and use the switch editor, I am able to assign different actions for buttons. However, the options are very limited, and for example, I MUST have "on" functionality on the same button as dim up, which is not what I want. Also, going through the switch editor here imposes a delay on switch presses. If I go to the switch group and assign lights instead, the functionality is pretty much what I want and the response is instant.

An example of what I want is:
From 21:00 - 07:00, when motion sensor detects motion, turn on "Night" scene. If associated wall switch on/off button is pressed, leave the light on until turned off from the wall switch again. 

I assume there would be an issue with the fact that the sensor rule would turn on the light, and if I pressed on/off, it would actually turn it off, and have me turn it on again, which could lead to annoying behaviour due to lost clicks if the button is pressed multiple times etc.

Is all this doable?

As I understand, there are a lot of possibilities by going outside of the Phoscon web interface in these senses already.  I tried to look into it, but I must admit, I have no idea where to start with the REST-API. Like @balu- is asking, it would be super nice, if there were a collection of examples to go by to start setting this up. Is that available something?

Side question, if some of you should know... Is there an iOS app that works? I haven't found a Phoscon app in the app store, and the web interface is broken on my iPhone. It crops out almost everything, so cannot use it for much of anything :(

Wyphorn, 2020-02-29:
> The motion sensor configuration will get a major refactor this year. It will include way more options to specify the time range as well as selecting scenes to trigger. Unfortunately this is not around the corner. Currently rules created with the REST-API are the only option.

Is here any news for the sensor configuration refactor? I hoped for it for the elvis release, but there is nothing in the release notes.
Is there alternatively some desktop app to configure rules to insert them with REST?

DerGute, 2020-03-04:
Thats interesting for me, too! Im not sure, whether the app hue essentials works that properly as it should. A native solution would be great!

Wyphorn, 2020-03-08:
@manup is there a vision for an official response?