dresden-elektronik/deconz-rest-plugin #431: Connect to HUE Dimmer Switch and E27 Bulb

nick4vit, 2018-02-22:
Hi all

I am not sure if I'm doing everything correctly: My conbee is connected to my windows machine (testing) and is working with different Xiaomi sensors. I also am able to add the Philips Dimmer Switch by resetting it (10s on the button, wait until its finished). But then something strange happens: When I am not connecting to the conbee an orange LEDs start to light up every few seconds and when I press the on button for several seconds, the bulb starts blinking and the Dimmer Switch gets connected. However if I am doing the same procedure but near the conbee it gets found in the webapp but the orange LED is not blinking. So I am not able to control the light using the Dimmer Switch as soon as it is coupled with the conbee.

According to the web ui in phoscon and webapp I am doing everything as it should be. Any idea? Should it be possible to control the bulb using the Dimmer Switch and conbee at the same time? If yes: could someone point out the procedure again so I can check if everything is done correctly? Thank you all for your help!

I'm using 2.04.99.

ebaauw, 2018-02-22:
Yes this is possible.  Just pair the dimmer switch with deCONZ.  Then add the lights to the group created by deCONZ, using the web app or the REST API.  I use this as a backup to control my lights while deCONZ is down.

nick4vit, 2018-02-22:
Thanks for your help! Can you quickly explain how the "add the lights" part is done? Thank you!

ebaauw, 2018-02-22:
In the web app, you should see the Hue dimmer switch under _Devices_.  When you click on the details (three dots), you get a window where you can change the names of the switch and the associated group, and where you can add lights to the group.

nick4vit, 2018-02-22:
Switch is there - yes. I click on the dots and "Add Lights" next to the "Group". But then there is an empty list, nothing to see, nothing to add? Do I need to do anything with the bulb?

ebaauw, 2018-02-22:
Do you see any lights in the main screen?

nick4vit, 2018-02-22:
No.

Here is what I did:

- Reset Dimmer Switch
- Connect it to deConz
-> RWL021 available in Main screen

When I click on "Add Lights" -> Nothing. No lights in Main screen.

When I go to Settings and Click on "Device Scan" it lists the light, I can click "Blink", which works and I can reset the device but then I lose the connection to the Dimmer Switch.

I also clicked on "Open Network" and restarted the bulb, nothing.

When I click on "Reset", "Open Network" and reboot the bulb, it is found as LWB010 and I can control the bulb via Webapp - but NOT from the Dimmer Switch anymore.

So the combo of being able to control the bulb from Webapp and Dimmer Switch is not possible by now and I have no clue how to fix it ;)

ebaauw, 2018-02-22:
Each time you pair the dimmer to a light directly, both will lose connection to deCONZ.

You need to pair both the light and the dimmer switch to deCONZ, and make sure deCONZ works with both.  Best delete the switch from the web app, reset it (push the hole for 10s, until it blinks), move the dimmer next to the ConBee, open the network from the web app, and wait for the dimmer to stop blinking and be recognised by deCONZ.  Double-check through the REST API or the web socket that deCONZ changes the `state.buttonevent` and `state.lastupdated` when you press a button.  
Then, in the web app, add the light to the group associated with the dimmer.

nick4vit, 2018-02-22:
I did exactly as you asked.

- Removed everything with Phoscon app
- Went to the other Webapp, Reseted Dimmer Switch (green light), Open Network
- After a while the Switch gets detected but no Light
- Switch works (Updates "Last Seen" in Desktop app when I hit buttons)

So how do I pair the bulb with deconz?

ebaauw, 2018-02-22:
Reset the bulb, power-cycle the bulb, open the network from the web app.

To reset a Hue bulb, hold the dimmer switch close to it and press and hold the On and Off buttons for ~10 seconds (wait for the light to react, and release after the light stops blinking).

nick4vit, 2018-02-23:
So you are suggesting:

1. Add Dimmer Switch as described above
2. Reset bulb by resetting/power-cyclng an opening the network

And then the light should be found by deconz and at the same time still work with the Dimmer Switch? I'm not home yet but I'm pretty sure that I have tested this already and as soon as I reset the bulb it loses the connection to the Dimmer Switch!

nick4vit, 2018-02-24:
I finally got this to work! Thank you @ebaauw for all your help!

closed 2018-02-24 by nick4vit

spridarn, 2018-06-19:
I use this thread for a similair problem. I have a E14 Hue lamp and a Hue dimmer. I bought the lamp and dimmer and first paired the dimmer to the lamp and it worked perfect. Today I bought a ConBee USB Zigbee-controller and connecte dit to my RPI running hass.io. I have installed deCONZ and first reseted the bulb. In the Phoscon web app I searched for the lamp and found it and added a name on it. The bulb works with my Home Assistant instans now. 
Now I saw that it was possible to connect the Hue Dimmer also so I can use Home Assistant or the Hue Dimmer to power on and off the lamp. The first thing I did was resetting it and the is was presented in the Phoscon web App. 
https://i.imgur.com/VY3aNPg.png
Then I added the bulb to the group just created for the dimmer as you can see in the picture below.
https://i.imgur.com/fS19Pui.png

But still nothing happens when I press the power on or off button on the dimmer. Any idea how to solve this? I tried to reset the bulb after adding the dimmer as the post also says but no success.

manup, 2018-06-19:
Now that both devices are joined to the network, you may try to hold the remote close to the bulb and Touchlink them (press `I`button several seconds).

spridarn, 2018-06-19:
@manup like this? https://youtu.be/9dPekfp4uJs
Nothing happens as you can see

manup, 2018-06-25:
Exactly, good video I'll keep the link if the issue pops up again :)

ebaauw, 2018-06-25:
@manup, didn't we tell the Hue dimmer switch not to touchlink?
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/1cc9e0f80670a42d8498a1737c3feae2c45b8f0d/de_web_plugin.cpp#L11238-L11241

Maybe the dimmer only configures the group after it's been told the magic word?  And the write attribute somehow got delayed?

manup, 2018-06-25:
Hmm good question might be worth a test.

Since the problems occurred only recently there might be also changes in the firmware. When I recall correctly Philips Hue bridge does only use the manufacturer specific command from the remote to bridge and then the bridge will control the lights based on rules.

ebaauw, 2018-06-25:
> When I recall correctly Philips Hue bridge does only use the manufacturer specific command from the remote to bridge and then the bridge will control the lights based on rules.

When paired to the Hue bridge: yes.  I got the "magic word" to attribute 0x0031 from sniffing the pairing and setup of the Hue dimmer by the Hue bridge.

We're now having this conversation over two topics, but are you sure the group ID is assigned by the Hue dimmer switch, or is it by the REST API plugin and the dimmer only knows about it through the (group) binding for the _OnOff_ and _Level_ cluster?  The Hue dimmer definitely works differently from any other dimmer switch I've seen.

spridarn, 2018-06-25:
@ebaauw for some reason I can´t see the version of the dimmer Phoscon web App now so can´t give you the version my dimmer has

manup, 2018-06-25:
>We're now having this conversation over two issues, but are you sure the group ID is assigned by the Hue dimmer switch, or is it by the REST API plugin and the dimmer only knows about it through the (group) binding for the OnOff and Level cluster? The Hue dimmer definitely works differently from any other dimmer switch I've seen.

Good question the binding might cause the effect that the group id is actually used. However as mentioned in the other thread, the id itself is assigned by the switch itself. On the other hand it might be totally possible to specify a own group ids via the binding. This is not possible with all switches, for example JUNG, GIRA and dresden elektronik switches chose their own group ids which can't be changed. Busch-Jaeger switches are free to be configured via bindings.

Switches are a mine field ;)

carloscae, 2019-01-04:
Was researching on this cause my Aqara switches are not really reliable. Hue Dimmer has a weird format for EU standards. What is the best option? Something that would talk directly with Hue bulbs?

timfosse, 2019-05-17:
Tried every suggestion in this thread but can not get my HUE Dimmer to turn on/off the bulbs associated with it. I can see the state changes in the websocket but nothing happens with the bulbs...