dresden-elektronik/deconz-rest-plugin #433: deCONZ / Conbee doesn't join

drbytes, 2018-02-23:
This morning I noticed a problem with deCONZ (latest beta), the Conbee can't seem to join the network:
[image]

Any idea why?

Is there a way to specify the tty port?

manup, 2018-02-23:
The tty port looks fine. since this a remote desktop session please make sure only one instance of deCONZ is running. Some clients will just open another X session where deCONZ from the other local session is not visible.

`$ ps ax | grep deCONZ`

If that isn't the problem please provide the settings from `Edit/Network Settings`.

drbytes, 2018-02-23:
No its not the right port, its on ttyUSB1
Ill test without the RFXCOM (ttyUSB0) disconnected

manup, 2018-02-23:
Hmm strange, then I don't understand how this screen is possible, do you have another FTDI device at ttyUSB0?

To disable auto connecting to a device the commandline parameter `--auto-connect=0` can be specified.
Then on the deCONZ start screen the device can be selected manually.

drbytes, 2018-02-24:
Yes, its another FTDI device, the RFXCOM. It seems there has been an issue with the data storage. I currently am looking at a total destruction of data. My backup seems corrupted as well.

I am now on a fresh install Is there any way to re-discover the devices? I have a ZLL.DB file that looks OK but no devices are shown in the UI. All the remotes still turn the lights on / off / dim. I just don't have anything in the UI. PWA shows all the devices but notes them as disconnected/unreachable.

Any way to recover without too much work?

manup, 2018-02-24:
:/ if you don't have an backup made in the Phoscon App it's difficult since the related network parameters are kept there.
It might be possible to recover since most parameters are also stored on the ConBee, is deCONZ in **In Network** state again?

If so you can see where your ZigBee network is:
* Open the old `WebApp` from deCONZ
* Go to  *Menu > Settings*
* At the bottom click `Scan for Devices`, this will tell you the network id and channel of the network.

In the Phoscon App you can switch to that channel:

*Menu > Settings > Gateway > Advanced*

**Note** This will only work if all other parameters are still ok.

drbytes, 2018-02-24:
[zll.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/1754462/zll.zip)
Here's my ZLL.DB

I've set the network and restarted deconz but still nothing
But I think I'm missing settings in network...
[image]

manup, 2018-02-24:
Uff it looks totally broken, which firmware version is shown in: *deCONZ > Help > About deCONZ* ?

drbytes, 2018-02-24:
The latest firmware I believe

[image]

drbytes, 2018-02-24:
I found another file in a partial backup, anything usefull here maybe
[ { "ExtAddress" : "0x00212effff01bb15", "NodeDescriptor" : "EECONRFHLQBBAC0AAA==", "NwkAddress" : "0x0000", "SceneX" : -249.64, "SceneY" : -796.933, "SimpleDescriptors" : [ "AQQBBQABARkAAQAF", "UADeAQABAAA=" ], "UserDescriptor" : "" }, { "ExtAddress" : "0x000b57fffe9b5c47", "NodeDescriptor" : "AUCOfBFSUgAAAFIAAA==", "NwkAddress" : "0x4d59", "SceneX" : 778.551, "SceneY" : -604.109, "SimpleDescriptors" : [ "AV7AAAICCQAAAwAEAAUABgAIAAADBQsAEAQFABkAIAAAEA==" ], "UserDescriptor" : "Bathroom Planet" }, { "ExtAddress" : "0x000b57fffebdc581", "NodeDescriptor" : "AUCOfBFSUgAAAFIAAA==", "NwkAddress" : "0x9bb0", "SceneX" : -76.0, "SceneY" : 75.0, "SimpleDescriptors" : [ "AV7AAAECCAAAAwAEAAUABgAIAAULABAEBQAZACAAABA=" ], "UserDescriptor" : "Bathroom shower" }, { "ExtAddress" : "0x000b57fffeb7a376", "NodeDescriptor" : "AUCOfBFSUgAAAFIAAA==", "NwkAddress" : "0xc32d", "SceneX" : -142.0, "SceneY" : 0.0, "SimpleDescriptors" : [ "AV7AAAECCAAAAwAEAAUABgAIAAULABAEBQAZACAAABA=" ], "UserDescriptor" : "Bathroom mirror" }, { "ExtAddress" : "0x00178801034858e9", "NodeDescriptor" : "AUCOCxBHLQAAAC0AAA==", "NwkAddress" : "0x1e1c", "SceneX" : 933.393, "SceneY" : -89.9825, "SimpleDescriptors" : [ "C17AAAECBwAAAwAEAAUABgAIAAAQARkA", "8uChYQAAASEAASEA" ], "UserDescriptor" : "Living Lamp" }, { "ExtAddress" : "0x000b57fffebfd6d1", "NodeDescriptor" : "AUCOfBFSUgAAAFIAAA==", "NwkAddress" : "0x423b", "SceneX" : -293.0, "SceneY" : -58.0, "SimpleDescriptors" : [ "AV7AAAECCAAAAwAEAAUABgAIAAULABAEBQAZACAAABA=" ], "UserDescriptor" : "Bar11" }, { "ExtAddress" : "0x000b57fffeb2c46b", "NodeDescriptor" : "AUCOfBFSUgAAAFIAAA==", "NwkAddress" : "0x4d41", "SceneX" : 127.369, "SceneY" : -696.383, "SimpleDescriptors" : [ "AV7AIAICCQAAAwAEAAUABgAIAAADBQsAEAQFABkAIAAAEA==" ], "UserDescriptor" : "Living Spot" }, { "ExtAddress" : "0x00178801103cebe8", "NodeDescriptor" : "AUCOCxBHLQAAAC0AAA==", "NwkAddress" : "0x9749", "SceneX" : 309.354, "SceneY" : -545.473, "SimpleDescriptors" : [ "C17AEAICCQAAAwAEAAUABgAIAAADABAB/AEZAA==", "8uChYQAAASEAASEA" ], "UserDescriptor" : "Wall Front" }, { "ExtAddress" : "0x00178801103ce8a0", "NodeDescriptor" : "AUCOCxBHLQAAAC0AAA==", "NwkAddress" : "0x4406", "SceneX" : 372.619, "SceneY" : 194.646, "SimpleDescriptors" : [ "C17AEAICCQAAAwAEAAUABgAIAAADABAB/AEZAA==", "8uChYQAAASEAASEA" ], "UserDescriptor" : "Wall Middle" }, { "ExtAddress" : "0x000b57fffeb84b06", "NodeDescriptor" : "AUCOfBFSUgAAAFIAAA==", "NwkAddress" : "0x6b7f", "SceneX" : -92.0, "SceneY" : 97.0, "SimpleDescriptors" : [ "AV7AAAECCAAAAwAEAAUABgAIAAULABAEBQAZACAAABA=" ], "UserDescriptor" : "Bar4" }, { "ExtAddress" : "0x000b57fffebdb1cd", "NodeDescriptor" : "AUCOfBFSUgAAAFIAAA==", "NwkAddress" : "0xa21d", "SceneX" : -218.0, "SceneY" : 355.0, "SimpleDescriptors" : [ "AV7AAAECCAAAAwAEAAUABgAIAAULABAEBQAZACAAABA=" ], "UserDescriptor" : "Bar2" }, { "ExtAddress" : "0x00178801103ce04a", "NodeDescriptor" : "AUCOCxBHLQAAAC0AAA==", "NwkAddress" : "0xa521", "SceneX" : 509.782, "SceneY" : 455.09, "SimpleDescriptors" : [ "C17AEAICCQAAAwAEAAUABgAIAAADABAB/AEZAA==", "8uChYQAAASEAASEA" ], "UserDescriptor" : "Wall Garden" }, { "ExtAddress" : "0x00178801001db826", "NodeDescriptor" : "AUCOCxBQUAAAAFAAAA==", "NwkAddress" : "0xd58e", "SceneX" : -765.0, "SceneY" : 363.0, "SimpleDescriptors" : [ "C17AAAICCQAAAwAEAAUABgAIAAADABAB/AEZAA==", "8uChYQAAASEAASEA" ], "UserDescriptor" : "Hue Bloom" }, { "ExtAddress" : "0x00178801105934df", "NodeDescriptor" : "AUCOCxBHLQAAAC0AAA==", "NwkAddress" : "0xdd94", "SceneX" : -779.968, "SceneY" : 559.979, "SimpleDescriptors" : [ "C17AAAECBwAAAwAEAAUABgAIAAAQARkA", "8uChYQAAASEAASEA" ], "UserDescriptor" : "Hall 1" }, { "ExtAddress" : "0x0017880110590c2a", "NodeDescriptor" : "AUCOCxBHLQAAAC0AAA==", "NwkAddress" : "0x1ba7", "SceneX" : -781.669, "SceneY" : 463.402, "SimpleDescriptors" : [ "C17AAAECBwAAAwAEAAUABgAIAAAQARkA", "8uChYQAAASEAASEA" ], "UserDescriptor" : "Hall 2" }, { "ExtAddress" : "0x000b57fffebf25e3", "NodeDescriptor" : "AUCOfBFSUgAAAFIAAA==", "NwkAddress" : "0x3f27", "SceneX" : -355.0, "SceneY" : -619.0, "SimpleDescriptors" : [ "AV7AAAECCAAAAwAEAAUABgAIAAULABAEBQAZACAAABA=" ], "UserDescriptor" : "Bathroom door" }, { "ExtAddress" : "0x000b57fffee8aa8c", "NodeDescriptor" : "AUCOfBFSUgAAAFIAAA==", "NwkAddress" : "0x7113", "SceneX" : 1029.79, "SceneY" : 219.238, "SimpleDescriptors" : [ "AV7AIAICCQAAAwAEAAUABgAIAAADBQsAEAQFABkAIAAAEA==" ], "UserDescriptor" : "Piano" }, { "ExtAddress" : "0x0017880102634d6b", "NodeDescriptor" : "AUCOCxBHLQAAAC0AAA==", "NwkAddress" : "0xfb7f", "SceneX" : 682.0, "SceneY" : -770.0, "SimpleDescriptors" : [ "C17AEAICCQAAAwAEAAUABgAIAAADABAB/AEZAA==", "8uChYQAAASEAASEA" ], "UserDescriptor" : "Bedroom" }, { "ExtAddress" : "0x001788010290bb91", "NodeDescriptor" : "AUCOCxBHLQAAAC0AAA==", "NwkAddress" : "0x1e13", "SceneX" : -492.0, "SceneY" : 98.0, "SimpleDescriptors" : [ "C17AEAICCQAAAwAEAAUABgAIAAADABAB/AEZAA==", "8uChYQAAASEAASEA" ], "UserDescriptor" : "Small Lamp" }, { "ExtAddress" : "0x000b57fffebe2c57", "NodeDescriptor" : "AUCOfBFSUgAAAFIAAA==", "NwkAddress" : "0x21ba", "SceneX" : 5.0, "SceneY" : 121.0, "SimpleDescriptors" : [ "AV7AAAECCAAAAwAEAAUABgAIAAULABAEBQAZACAAABA=" ], "UserDescriptor" : "Bar13" }, { "ExtAddress" : "0x0017880102dc2bdf", "NodeDescriptor" : "AUCOCxBHLQAAAC0AAA==", "NwkAddress" : "0xd2bb", "SceneX" : 986.568, "SceneY" : -470.184, "SimpleDescriptors" : [ "C17AEAICCQAAAwAEAAUABgAIAAADABAB/AEZAA==", "8uChYQAAASEAASEA" ], "UserDescr [... cut ...] ch" }, { "ExtAddress" : "0x00178801030d8166", "NodeDescriptor" : "AUCOCxBHLQAAAC0AAA==", "NwkAddress" : "0x806b", "SceneX" : -733.327, "SceneY" : -397.304, "SimpleDescriptors" : [ "C17AAAECBwAAAwAEAAUABgAIAAAQARkA", "8uChYQAAASEAASEA" ], "UserDescriptor" : "Kitchen" }, { "ExtAddress" : "0x000b57fffe9ccb59", "NodeDescriptor" : "AkCAfBFSUgAAAFIAAA==", "NwkAddress" : "0x9a10", "SceneX" : -701.0, "SceneY" : -121.0, "SimpleDescriptors" : [ "AQQB//8AAQEAAA==" ], "UserDescriptor" : "Bedroom switch" } ]

drbytes, 2018-02-24:
That worked, however I had to move it to another machine, it doesn't want to work on the ubuntu machine.
After I reinstalled everything on a windows machine I pulled a backup in pwa, moved the stick to ubuntu, applied the backiup and it's back to "joining" and not doing anything. A bunch of CRC errors are also thrown by the application

manup, 2018-02-25:
Is this a plain Ubuntu machine or VM? Can you try another USB slot, and also try to just connect the ConBee without the other RFCCOM connected.

CRC errors indicate that multiple programs communicate via UART with ConBee.

Also since you connect remotely to the Ubuntu machine have you verified that only one X session is running, CRC errors also occur when multiple deCONZ instances are running.

Can you please provide the output of:

    $ ps ax | grep deCONZ

closed 2019-03-30 by stale[bot]

autopia40, 2024-02-20:
Unsure if this will help anyone that comes across this thread, I found it as I couldn't get a ConBee 1 and DeConz to join and map an existing zigbee network, after playing with all the Deconz settings, I think I found the correct settings, which were essentially two parameters over and above setting it to router; I enabled predefined PAN ID and filled in the PAN ID field from my zigbee network, and Security mode only worked with 'High - No Master but TC Link Key', the address and key fields were autopopulated on joining the network.  I also had to find out the channel being used and selected it in the Channel Mask, hope this all helps someone trying to use a ConBee device and DeConz?

[image]
[image]