dresden-elektronik/deconz-rest-plugin #434: NEO Coolcam / Everspring SP816 - Compability

bjoernhoefer, 2018-02-24:
Hi there, 

does anyone have some experience with the NEO Coolcam Motion Sensor?

[Vendor Documenation](http://www.szneo.com/en/products/show.php?id=190)

Otherwise I was thinking about the [Everspring SP816](http://www.everspring.com/portfolio-item/sp816-outdoor-motion-detector/)

Is it supported - as they uses Z-Wave Plus - is this supported by deCONZ/pwa and the raspbee module?

Thanks for your support

Björn

mysteryan, 2018-02-24:
deCONZ/pwa with raspbee/conbee is zigbee orientated.  On top of that, zwave operates at a different RF spectrum.  Refer to OpenZWave for zwave devices.

manup, 2018-02-24:
As noted Z-Wave devices can't be supported by deCONZ since the underling hardware ConBee/RaspBee is ZigBee only which is a different radio standard than Z-Wave.

You may have a look at the RaZberry shield which is like RaspBee but for Z-Wave. Note: Be aware it uses a own software stack and has no integration in deCONZ/Phoscon.

https://new.z-wave.me/products/razberry/

closed 2019-03-30 by stale[bot]