dresden-elektronik/deconz-rest-plugin #435: Conbee / ESXI / No connections

nick4vit, 2018-02-24:
Hi all

In the past couple of days I was testing deconz/conbee on my Windows workstation. That worked pretty well. As I have an ESXI host I wanted to host the deconz software on a virtual machine. Passthrough works, Conbee stick gets recognized (I also tested in a Windows VM) and is used - but it is not possible to get any connections to devices.

So I was wondering if:

- Is a special "reset" of the stick needed when I use it on a different machine?
- How can I debug, if the stick actually is sending and receiving data from other devices?
- Is the conbee not working using USB passthrough?

Thanks for some ideas!

ebaauw, 2018-02-24:
I was able to run deCONZ on a Windows7 VM (and BitCatcher on a Ubuntu VM) on my MacBook using Parallels.  I had to connect the ConBee directly to the MacBook; when connected through a USB hub, it's recognised but didn't work.

nick4vit, 2018-02-24:
So you did not have to reset anything special when moving it around? As I see a lot of "action" in the log file I think the stick is working properly. But I don't know why it doesn't find any other devices. Could the configuration within deconz (channels etc.) be a problem?

manup, 2018-02-24:
No there is no special handling needed I also switch ConBees between various VMs including Windows all the time.

In Virtualbox the USB machine settings must include USB 2.0 but that's it basically.

nick4vit, 2018-02-24:
Okay! Is there a deconz debug option which could help? When I start deconz on ubuntu with debug on it looks like that:

`12:13:05:021 COM: /dev/ttyUSB0 : FT230X Basic UART
12:13:05:028 COM: /dev/ttyUSB0 : FT230X Basic UART
12:13:05:028 auto connect com /dev/ttyUSB0
12:13:05:665 Serial com connected
12:13:05:683 int zmMaster::readParameter(ZM_DataId_t) 0x22
12:13:05:684 int zmMaster::readParameter(ZM_DataId_t) 0x01
12:13:05:684 int zmMaster::readParameter(ZM_DataId_t) 0x05
12:13:05:684 int zmMaster::readParameter(ZM_DataId_t) 0x08
12:13:05:684 int zmMaster::readParameter(ZM_DataId_t) 0x07
12:13:05:684 int zmMaster::readParameter(ZM_DataId_t) 0x0A
12:13:05:684 int zmMaster::readParameter(ZM_DataId_t) 0x09
12:13:05:684 int zmMaster::readParameter(ZM_DataId_t) 0x0E
12:13:05:684 int zmMaster::readParameter(ZM_DataId_t) 0x0D
12:13:05:684 int zmMaster::readParameter(ZM_DataId_t) 0x10
12:13:05:684 int zmMaster::readParameter(ZM_DataId_t) 0x0B
12:13:05:684 int zmMaster::readParameter(ZM_DataId_t) 0x15
12:13:05:684 int zmMaster::readParameter(ZM_DataId_t) 0x1C
12:13:05:684 int zmMaster::readParameter(ZM_DataId_t) 0x1E
12:13:05:684 int zmMaster::readParameter(ZM_DataId_t) 0x21
12:13:05:685 int zmMaster::readParameter(ZM_DataId_t) 0x24
12:13:05:685 int zmMaster::readParameter(ZM_DataId_t) 0x23
12:13:05:685 int zmMaster::readParameter(ZM_DataId_t) 0x1D
12:13:05:685 int zmMaster::readParameter(ZM_DataId_t) 0x20
12:13:05:685 int zmMaster::readParameterWithArg(ZM_DataId_t, uint8_t*, uint8_t) 0x18
12:13:05:685 int zmMaster::readParameterWithArg(ZM_DataId_t, uint8_t*, uint8_t) 0x13
12:13:05:685 int zmMaster::readParameterWithArg(ZM_DataId_t, uint8_t*, uint8_t) 0x13
12:13:05:685 int zmMaster::readParameter(ZM_DataId_t) 0x16
12:13:05:685 int zmMaster::readParameter(ZM_DataId_t) 0x25
12:13:05:685 int zmMaster::readParameter(ZM_DataId_t) 0x22
12:13:05:685 int zmMaster::readParameter(ZM_DataId_t) 0x01
12:13:05:685 int zmMaster::readParameter(ZM_DataId_t) 0x05
12:13:05:685 int zmMaster::readParameter(ZM_DataId_t) 0x08
12:13:05:685 int zmMaster::readParameter(ZM_DataId_t) 0x07
12:13:05:686 int zmMaster::readParameter(ZM_DataId_t) 0x0A
12:13:05:686 int zmMaster::readParameter(ZM_DataId_t) 0x09
12:13:05:686 int zmMaster::readParameter(ZM_DataId_t) 0x0E
12:13:05:686 int zmMaster::readParameter(ZM_DataId_t) 0x0D
12:13:05:686 int zmMaster::readParameter(ZM_DataId_t) 0x10
12:13:05:686 int zmMaster::readParameter(ZM_DataId_t) 0x0B
12:13:05:686 int zmMaster::readParameter(ZM_DataId_t) 0x15
12:13:05:686 int zmMaster::readParameter(ZM_DataId_t) 0x1C
12:13:05:686 int zmMaster::readParameter(ZM_DataId_t) 0x1E
12:13:05:686 int zmMaster::readParameter(ZM_DataId_t) 0x21
12:13:05:686 int zmMaster::readParameter(ZM_DataId_t) 0x24
12:13:05:686 int zmMaster::readParameter(ZM_DataId_t) 0x23
12:13:05:686 int zmMaster::readParameter(ZM_DataId_t) 0x1D
12:13:05:686 int zmMaster::readParameter(ZM_DataId_t) 0x20
12:13:05:686 int zmMaster::readParameterWithArg(ZM_DataId_t, uint8_t*, uint8_t) 0x18
12:13:05:686 int zmMaster::readParameterWithArg(ZM_DataId_t, uint8_t*, uint8_t) 0x13
12:13:05:687 int zmMaster::readParameterWithArg(ZM_DataId_t, uint8_t*, uint8_t) 0x13
12:13:05:687 int zmMaster::readParameter(ZM_DataId_t) 0x16
12:13:05:687 int zmMaster::readParameter(ZM_DataId_t) 0x25
This plugin does not support propagateSizeHints()
12:13:05:699 Device firmware version 0x26190500
12:13:05:731 unlocked max nodes: 200
12:13:05:731 has feature: max nodes (0x11), status: 0x00
12:13:05:747 has feature: std security (0x01), status: 0x00
12:13:05:763 has feature: link security (0x02), status: 0x00
12:13:05:779 has feature: high security (0x03), status: 0x04
12:13:05:795 has feature: ethernet (0x04), status: 0x04
12:13:05:811 has feature: mass storage device (MSD) (0x05), status: 0x04
12:13:05:827 has feature: device firmware upgrade (DFU) (0x06), status: 0x04
12:13:05:843 has feature: ZigBee Light Link (ZLL) (0x07), status: 0x04
12:13:05:859 has feature: interpan (0x08), status: 0x04
12:13:05:875 has feature: button 1 (0x09), status: 0x04
12:13:05:891 has feature: button 1 (0x0A), status: 0x04
12:13:05:907 has feature: LED 1 (0x0B), status: 0x00
12:13:05:923 has feature: LED 2 (0x0C), status: 0x00
[... 162 lines cut ...]
12:13:25:841 APS-DATA.request id: 20 erase from queue
12:13:25:841 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
12:13:26:207 Idle timer triggered
12:13:26:207 enable support for CIE 1931 XY color mode for groups
--dbg-info=3 --dbg-aps=112:13:28:119 APS-DATA.request id: 22, addrmode: 0x02, addr: 0x0000, profile: 0x0000, cluster: 0x0031, ep: 0x00 queue: 0 len: 2
12:13:28:210 APS-DATA.confirm id: 22, status: 0x00 SUCCESS
12:13:28:210 APS-DATA.confirm request id: 22 -> confirmed, timeout 59840
12:13:28:242 APS-DATA.indication srcAddr: 0x00212effff01b882, dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 30, rssi: -31
12:13:28:243 APS-DATA.indication request id: 22 -> finished
12:13:28:243 APS-DATA.request id: 22 erase from queue
12:13:28:243 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
12:13:30:518 APS-DATA.request id: 24, addrmode: 0x02, addr: 0x0000, profile: 0x0000, cluster: 0x0031, ep: 0x00 queue: 0 len: 2
12:13:30:612 APS-DATA.confirm id: 24, status: 0x00 SUCCESS
12:13:30:612 APS-DATA.confirm request id: 24 -> confirmed, timeout 59840
[... cut ...]
12:13:32:918 APS-DATA.request id: 26, addrmode: 0x02, addr: 0x0000, profile: 0x0000, cluster: 0x0031, ep: 0x00 queue: 0 len: 2
12:13:32:997 APS-DATA.confirm id: 26, status: 0x00 SUCCESS
12:13:32:998 APS-DATA.confirm request id: 26 -> confirmed, timeout 59920
12:13:33:029 APS-DATA.indication srcAddr: 0x00212effff01b882, dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 30, rssi: -31
12:13:33:030 APS-DATA.indication request id: 26 -> finished
12:13:33:030 APS-DATA.request id: 26 erase from queue
12:13:33:030 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000`

nick4vit, 2018-02-24:
One more log: I opened the network and resetted a Xiaomi Aqare Motion Sensor and then this happened, maybe one of you sees something strange? What about that channel change?

12:15:25:114 Device protocol version: 0x0105
12:15:25:130 CTRL restore cached node 0x00212effff01b882
12:15:25:130 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 2, node: 0x0000
12:15:25:130 new node - ext: 0x00212effff01b882, nwk: 0x0000
12:15:25:130 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 2, node: 0x0000
12:15:25:178 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 1, node: 0x0000
12:15:25:226 int zmMaster::readParameterWithArg(ZM_DataId_t, uint8_t*, uint8_t) 0x19
12:15:25:306 Current channel 25
12:15:25:370 CTRL ANT_CTRL 0x03
12:15:25:386 CTRL read param resp id: 0x1D, status: 0x04
12:15:25:402 CTRL read param resp id: 0x20, status: 0x04
12:15:25:514 Device protocol version: 0x0105
12:15:25:578 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 1, node: 0x0000
12:15:25:614 APS-DATA.request id: 2, addrmode: 0x02, addr: 0x0000, profile: 0x0000, cluster: 0x0031, ep: 0x00 queue: 0 len: 2
12:15:25:626 int zmMaster::readParameterWithArg(ZM_DataId_t, uint8_t*, uint8_t) 0x19
12:15:25:706 Current channel 25
12:15:25:770 CTRL ANT_CTRL 0x03
12:15:25:786 CTRL read param resp id: 0x1D, status: 0x04
12:15:25:802 CTRL read param resp id: 0x20, status: 0x04
12:15:26:042 APS-DATA.confirm id: 2, status: 0x00 SUCCESS
12:15:26:043 APS-DATA.confirm request id: 2 -> confirmed, timeout 59520
12:15:26:074 APS-DATA.indication srcAddr: 0x00212effff01b882, dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 30, rssi: -31
12:15:26:075 APS-DATA.indication request id: 2 -> finished
12:15:26:075 APS-DATA.request id: 2 erase from queue
12:15:26:075 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
12:15:28:015 APS-DATA.request id: 4, addrmode: 0x02, addr: 0x0000, profile: 0x0000, cluster: 0x0031, ep: 0x00 queue: 0 len: 2
12:15:28:124 APS-DATA.confirm id: 4, status: 0x00 SUCCESS
12:15:28:124 APS-DATA.confirm request id: 4 -> confirmed, timeout 59840
12:15:28:156 APS-DATA.indication srcAddr: 0x00212effff01b882, dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 66, rssi: -31
12:15:28:156 APS-DATA.indication request id: 4 -> finished
12:15:28:156 APS-DATA.request id: 4 erase from queue
12:15:28:156 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
12:15:28:714 COM: /dev/ttyUSB0 : FT230X Basic UART
12:15:28:714 GW firmware update select USB device
12:15:28:714 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_Rpi_0x26190500.bin.GCF
12:15:28:714 GW firmware version: 0x26190500
12:15:28:714 GW firmware version is up to date: 0x26190500
12:15:28:959 Announced to internet
12:15:30:414 APS-DATA.request id: 6, addrmode: 0x02, addr: 0x0000, profile: 0x0000, cluster: 0x0031, ep: 0x00 queue: 0 len: 2
12:15:30:509 APS-DATA.confirm id: 6, status: 0x00 SUCCESS
12:15:30:509 APS-DATA.confirm request id: 6 -> confirmed, timeout 59840
12:15:30:541 APS-DATA.indication srcAddr: 0x00212effff01b882, dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 30, rssi: -31
12:15:30:541 APS-DATA.indication request id: 6 -> finished
12:15:30:541 APS-DATA.request id: 6 erase from queue
12:15:30:542 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
12:15:32:814 APS-DATA.request id: 8, addrmode: 0x02, addr: 0x0000, profile: 0x0000, cluster: 0x0031, ep: 0x00 queue: 0 len: 2
12:15:32:927 APS-DATA.confirm id: 8, status: 0x00 SUCCESS
12:15:32:927 APS-DATA.confirm request id: 8 -> confirmed, timeout 59840
12:15:32:959 APS-DATA.indication srcAddr: 0x00212effff01b882, dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 30, rssi: -31
12:15:32:959 APS-DATA.indication request id: 8 -> finished
12:15:32:959 APS-DATA.request id: 8 erase from queue
12:15:32:959 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
12:15:33:706 channel 25 does not match channel mask 0x00000800 (TODO)
12:15:33:706 channel is 25 but should be 11, start channel change
12:15:33:706 ChannelChangeState: CC_verify_Channel
12:15:34:706 channel change NOT verified!
12:15:34:706 ChannelChangeState: CC_Change_Channel
12:15:35:214 APS-DATA.request id: 10, addrmode: 0x02, addr: 0x0000, profile: 0x0000, cluster: 0x0031, ep: 0x00 queue: 0 len: 2
12:15:35:312 APS-DATA.confirm id: 10, status: 0x00 SUCCESS
12:15:35:313 APS-DATA.confirm request id: 10 -> confirmed, timeout 59840
12:15:35:344 APS-DATA.indication srcAddr: 0x00212effff01b882, dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 30, rssi: -31
12:15:35:344 APS-DATA.indication request id: 10 -> finished
12:15:35:345 APS-DATA.request id: 10 erase from queue
12:15:35:345 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
12:15:35:706 change channel with nwkUpdateId = 11
12:15:35:706 APS-DATA.request id: 12, addrmode: 0x02, addr: 0xfffc, profile: 0x0000, cluster: 0x0038, ep: 0x00 queue: 0 len: 7
12:15:35:706 change channel to 11, channel mask = 0x00000800
12:15:35:707 ChannelChangeState: CC_WaitConfirm
[... 539 lines cut ...]
12:17:23:301 APS-DATA.confirm request id: 103 -> confirmed, timeout 59840
12:17:23:333 APS-DATA.indication srcAddr: 0x00212effff01b882, dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 30, rssi: -31
[... cut ...]
12:17:28:104 APS-DATA.confirm request id: 107 -> confirmed, timeout 59840
12:17:28:136 APS-DATA.indication srcAddr: 0x00212effff01b882, dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 30, rssi: -31
12:17:28:136 APS-DATA.indication request id: 107 -> finished
12:17:28:136 APS-DATA.request id: 107 erase from queue
12:17:28:136 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
^C12:17:29:052 shutdown after signal(2)
12:17:29:053 REST API plugin shutting down

nick4vit, 2018-02-24:
Wow! I switched the USB port - and all of a sudden its working!

ebaauw, 2018-02-24:
Maybe the other port was connected through an internal USB hub?

nick4vit, 2018-02-24:
I have a SuperMicro X10SAE motherboard. I just checked the manual and the ports I used are from the same USB 2.0 group. So there shouldn't be any difference between these two ports. Maybe something with the port isn't right. Strange.

nick4vit, 2018-02-24:
Very

manup, 2018-02-28:
I've only seen such problems when VMs were running and grabbed the device so the host had no more access to it.

nick4vit, 2018-02-28:
I found the problem: multiple gateway configurations on the same VM from testing with different ports. Once I've removed all gateway configurations I was able to set it up properly!

closed 2018-02-28 by nick4vit