dresden-elektronik/deconz-rest-plugin #439: eCozy thermostat valves

mshli, 2018-02-25:
Does anyone has some experience with eCozy thermostat valves? (ecozy.de) They work with ZHA and i want to use them with my Raspbee and deConz.

KingTomaHawk, 2018-02-27:
+1
The thermostats are very cheap at the moment and have a very cool design. If more device specific data is required, I could buy one and post some screenshots of the requested values in deconz.

mshli, 2018-02-27:
I have one for testing and it‘s allready connected. If someone needs some screenshots, just say. I have also send a request to the technical support of ecozy. Maybe they send me the required informations.

manup, 2018-03-01:
Discussion continues in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/209

closed 2018-03-01 by manup