dresden-elektronik/deconz-rest-plugin #440: Change description.xml to be identical to Hue Bridge?

FezVrasta, 2018-02-26:
Hi, I'm trying to pair my Philips Ambilight TV with my DeCONZ but it doesn't seem to get listed in the available bridges.

I can connect the Philips Hue app to my DeCONZ just fine so I guess the TV is simply more picky and doesn't like the `description.xml` that's different from the one of a real Hue Bridge.

Is there any chance to get a "stealth mode" to maximize compatibility?

ebaauw, 2018-02-26:
Are you sure the TV looks at `description.xml`?  Can you confirm by sniffing the network?  Far more likely is (in my opinion):
- The TV doesn't use UPnP to discover the bridge and only looks at the meethue portal (verify by disconnecting your router from the Internet and checking if the TV still finds the Hue bridge);
- The TV does use UPnP but checks for more than the Hue app, like the server signature, `FreeRTOS/7.4.2, UPnP/1.0, IpBridge/1.8.0` for deCONZ vs `Linux/3.14.0 UPnP/1.0 IpBridge/1.23.0` for the Hue bridge, or even the mac address prefix in `hue-bridgeid`

FezVrasta, 2018-02-26:
I think it's `description.xml` because my TV can see just fine the `emulated_hue` bridge of HASS

closed 2019-03-30 by stale[bot]