dresden-elektronik/deconz-rest-plugin #442: deCONZ gui crashes when reading "Present value" (float) in attribute editor

FrodoKenny, 2018-02-26:
When I add the Magic Cube to deCONZ and manually read the "Present value" of the "Analog Input (Basic)" cluster in Attribute Editor, the gui crashes and deCONZ restarts. This might have something to do with it being a float instead of an integer? I think it only happens when I just added the Magic Cube and the value reading succeeds. After the Magic Cube been added the read always seems to fail.

ebaauw, 2018-02-26:
I think that, like most Xiaomi devices, the cube doesn't respond to read attributes (except for the _Basic_ cluster).  It's based on attribute reporting, so the value in the GUI should update in real time when you rotate the cube.

FrodoKenny, 2018-02-27:
Thanks, indeed the value updates when rotating. I guess it still shouldn't crash though when manually trying to read...

I also noticed the read attributes from the _Basic_ cluster don't survive a reboot. Is that expected?

ebaauw, 2018-02-27:
I cannot reproduce the crash.

Yes, after retarting deCONZ, the data in the GUI is lost.  The REST API plugin persists everything in its database, but it just doesn’t show in the GUI.

FrodoKenny, 2018-02-27:
Hmm, it consistently happens for me, but it might be my setup. I'm running 2.05.07 on a raspberry pi 3, recently updated.

I need to reset and remove the cube first, then click add sensors in the Phoscon app and click the link button on the cube. If I then go to the deCONZ gui and read that value in attribute editor it crashes. Once the network isn't open anymore, it doesn't seem to crash anymore, so it might not be very important to fix.

In any case, I attached the last part of the log. The last line "12:29:26:435 void de" looks to be cut off when it crashed.

----

12:29:24:880 HTTP API PUT /api/C9AEAF08BC/config - 127.0.0.1
12:29:24:880 Text Data: 	{"permitjoin":0}
12:29:24:880 [{"success":{"/config/permitjoin":0}}]
12:29:24:893 HTTP client hdr detected:
GET /api/C9AEAF08BC/sensors/new HTTP/1.1

Host: 127.0.0.1

Connection: keep-alive

Accept: vnd.ddel.v1

X-Requested-With: XMLHttpRequest

User-Agent: Mozilla/5.0 (X11; Linux armv7l) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/60.0.3112.89 Safari/537.36

Referer: http://127.0.0.1/pwa/devices-sensors.html

Accept-Encoding: gzip, deflate, br

Accept-Language: en-GB,en-US;q=0.8,en;q=0.6

12:29:24:893 HTTP API GET /api/C9AEAF08BC/sensors/new - 127.0.0.1
12:29:24:893 {"1":{"name":"lumi.sensor_cube 1"},"2":{"name":"lumi.sensor_cube 2"},"lastscan":"active"}
12:29:25:165 Node 0x00158D00010DF2EF is known by 1 neighbors, last seen 2 s
12:29:25:468 HTTP client hdr detected:
GET /api/C9AEAF08BC/sensors HTTP/1.1

Host: 127.0.0.1

Connection: keep-alive

Accept: vnd.ddel.v1

X-Requested-With: XMLHttpRequest

User-Agent: Mozilla/5.0 (X11; Linux armv7l) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/60.0.3112.89 Safari/537.36

Referer: http://127.0.0.1/pwa/devices-sensors.html

Accept-Encoding: gzip, deflate, br

Accept-Language: en-GB,en-US;q=0.8,en;q=0.6

If-None-Match: "8ff313faa0ae77161d90123c64393f5b"

12:29:25:468 HTTP API GET /api/C9AEAF08BC/sensors - 127.0.0.1
12:29:25:644 Node 0x00158D0001E533AA is known by 0 neighbors, last seen 17 s
12:29:26:124 Node 0x00158D00010DF2EF is known by 1 neighbors, last seen 2 s
12:29:26:226 MAC Poll 0x02 0x5DC0
12:29:26:243 APS-DATA.confirm id: 88, status: 0x00 SUCCESS
12:29:26:243 APS-DATA.confirm request id: 88 -> erase from queue
12:29:26:284 aps request id: 88 finished, erase from queue
12:29:26:365 MGTM_Lqi_req zdpSeq: 23 to 0x00212EFFFF0140EE start index 0
12:29:26:365 APS-DATA.request id: 89, addrmode: 0x02, addr: 0x0000, profile: 0x0000, cluster: 0x0031, ep: 0x00 queue: 0 len: 2
12:29:26:393 APS-DATA.confirm id: 89, status: 0x00 SUCCESS
12:29:26:393 APS-DATA.confirm request id: 89 -> confirmed, timeout 59920
12:29:26:411 APS-DATA.indication srcAddr: 0x00212effff0140ee, dstAddrMode: 2, profile: 0x0000, cluster: 0x0031, lqi: 255, rssi: 0
12:29:26:411 ZDP status = 0x00 -> SUCCESS
12:29:26:435 APS-DATA.indication srcAddr: 0x00212effff0140ee, dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 255, rssi: 0
12:29:26:435 APS-DATA.indication request id: 89 -> finished
12:29:26:435 APS-DATA.request id: 89 erase from queue
12:29:26:435 ZDP status = 0x00 -> SUCCESS
12:29:26:435 void de

closed 2019-03-30 by stale[bot]