dresden-elektronik/deconz-rest-plugin #443: openhab2 Binding

steveknoefler, 2018-02-27:
Hi
could you PLEASE develop a binding for openhab which using web sockets for eg. Motionsensors updates?

thanks
Steve

manup, 2018-02-27:
Hi, you may checkout the openhab ZigBee binding, there was some early support for the ConBee/RaspBee(?) but I'm not aware of the current state. The development seems quite active so it might be the best bet for ZigBee in openhab.

https://github.com/openhab/org.openhab.binding.zigbee

Kane610, 2018-02-27:
Or you could come over to Home assistant, it works great here 👍

sheineme, 2018-02-27:
I'm using the Hue Binding with deConz. It does most of the stuff i need; what is missing is recalling scenes, for this I use the HTTP binding to call deConz REST API directly

steveknoefler, 2018-02-27:
yes thats right but the polling every seconds is to slow for motion sensors.

steveknoefler, 2018-02-27:
I checked home assistant but there is no lux in motion sensors properties :/

manup, 2018-02-27:
> I checked home assistant but there is no lux in motion sensors properties :/

You may ask the developers of the binding if this feature is planned or already existing somehow.

steveknoefler, 2018-02-27:
I checked it with my own installation

steveknoefler, 2018-02-27:
oh I missed it :(

Kane610, 2018-02-27:
@steveknoefler lux is reported as its own sensor from motion sensors

radi-ka, 2018-03-24:
Hi,

i just started working on a binding that connects to deCONZ via websockets. Right now it only supports temperature, humidity and open/closed sensors in a rudimentary way (only state updates, no update time, no battery level etc).

If you want to have a look:
[https://github.com/radi-ka/deconz_oh2_binding](https://github.com/radi-ka/deconz_oh2_binding)

steveknoefler, 2018-03-25:
sorry for late. thats what I see on api for my motionsensor: (deconz api)

{"config":{"alert":"none","battery":100,"ledindication":false,"on":true,"pending":[],"reachable":true,"tholddark":12000,"tholdoffset":7000,"usertest":false},"ep":2,"etag":"86fa3883746c8c6c6ccf6dd173ffbf63","manufacturername":"Philips","modelid":"SML001","name":"FlurOG_Bewegungsmelder_2","state":{"dark":true,"daylight":false,"lastupdated":"2018-03-25T18:04:51","lightlevel":8417,"lux":7},"swversion":"6.1.0.18912","type":"ZHALightLevel","uniqueid":"00:17:88:01:02:13:10:87-02-0400"}

TwE3d, 2018-05-20:
@sheineme @steveknoefler Can you give any Explanation how you got the motion sensor working in openhab? When i wanted to add the sensor manually there is only the possibility to add lights, not motion sensors... Im kinda stuck right now...

goermezer, 2018-05-31:
Same for me. I need to integrate motion and door sensors from Xiaomi, which work quite well in deCONZ. But I couldn`t find a way to bring them to OpenHAB. A binding would be really perfect.

manup, 2018-05-31:
It might be worth reaching out to the OpenHAB community 

https://community.openhab.org/t/raspbee-binding-compatibility/10020

renegade2k, 2018-07-05:
@radi-ka are you still working on it?
I'm looking for a way to get the Ikea Tradfri motion sensor to work with openhab somehow, without going through the Tradfri Gateway

radi-ka, 2018-07-05:
@renegade2k sort of. The current state on github is what i am running at home and what fills my (current) needs.

Assuming tradfri motion sensors emit a websocket event like this: `{"e":"changed","id":"9","r":"sensors","state":{"lastupdated":"2018-03-18T20:26:21","presence":true},"t":"event"}`
they should work right now.

If you need help configuring the plugin, please open an issue there.

sheineme, 2019-01-11:
There is now also an openhab2 binding available in the "official" repository.
https://www.openhab.org/addons/bindings/deconz/

closed 2019-05-18 by stale[bot]