dresden-elektronik/deconz-rest-plugin #444: Allow to specify a single IP address to use

FezVrasta, 2018-02-27:
Since I'm running PiHole and DeCONZ on the same Raspberry Pi 3, I setup a virtual interface (`eth0:0`) with its own IP address.

I configured PiHole to run on the virtual interface, but DeCONZ will always listen on both the IPs preventing PiHole from starting.

Could you add an option to specify which IP to use?

manup, 2018-02-28:
With upcoming deCONZ version 2.05.09 the listen interface can be specified on command line and config file. The special value of 0.0.0.0 means listen on any interface.

```
--http-listen=192.168.70.4
```

or persistant via config.ini:

```
[http]
port=80
listen=0.0.0.0
...
```

FezVrasta, 2018-02-28:
Great! Is there any approx ETA already maybe?

manup, 2018-02-28:
For sure this week, not so sure which day :)

manup, 2018-03-04:
You may checkout version 2.05.10, the above mentioned command line options are included now.

FezVrasta, 2018-03-04:
I tried to edit the deconz.service and added the `--http-listen` argument but it seems to get ignored.

May you point me to the right way to configure it please?

FezVrasta, 2018-03-04:
Oh... weirdly now the http://www.dresden-elektronik.de/pwa app won't discover my DeCONZ instance.

The self-hosted web interface works just fine tho.

closed 2018-03-04 by FezVrasta

FezVrasta, 2018-03-04:
Neither Home Assistant will be able to discover it, and if I set the IP manually it won't allow me to pair it.

FWillems, 2018-07-22:
@FezVrasta
Have you been able to run two webserver on one raspberry?
I'm trying to get it working to have
1. pimatic (home automation webserver) on port 80 by wlan0 with IP 192.168.2.100 and
2. the raspbee module with its webserver also on port 80 by wlan0:0 with IP 192.168.2.101

Is that possible? Can you explain your solution?
Thank you.

FezVrasta, 2018-07-22:
You should be able to use the cli argument mentioned above

DurrM, 2019-03-11:
So after I installed DeCONZ to my Raspberry I was no longer able to access the admin page of PiHole. I would like to keep PiHole on eth0 interface and "move" deconz to the virtual interface eth0:0. I'm just not sure if that's possible and how to do it. 
Sorry but I'm pretty new. Hope somebody can help me or at least give me some link with further information.

Thank you

airbone42, 2020-06-06:
Sorry for bringing this old thing up again, but I'm trying to use the --http-listen parameter on a Windows 7 machine with deCONZ 2.05.75.

While the --http-port parameter works perfectly it seems to ignore --http-listen. Even when specified the REST plugin and Phoscon web app are available on all ip-addresses, not just the one defined. As I'm running openhab in parallel and want to use the hue emulation form both systems I have to specify a single ip address, to avoid conflicts.

I wanted to try the config.ini setting as well, but couldn't find any information where to find/place the config.ini on a Windows-system.

joggs, 2020-08-06:
Cannot find config.ini either. Please specify

Mimiix, 2020-08-06:
Its in the ZLL.db afaik. However, normally this is something that is done in the router or host OS as deCONZ isn't a network adapter.

CapitalT, 2021-03-19:
Proper way to change service file s it survives updates is to use `sudo systemctl edit deconz`. First empty ExecStart before setting new one.
See: https://github.com/dresden-elektronik/phoscon-app-beta/issues/26#issuecomment-410553910