dresden-elektronik/deconz-rest-plugin #446: API usage without API key

Framspott, 2018-02-28:
Hi everyone,

I want to integrate the REST plugin into a software project to connect to Zigbee devices. I was able to use your API successfully, but the API key generation is not very comfortable. I simply want to use your interface without the need of the web applikation and without opening the gateway for generating a API key.

Therefore I want to know if there is a way to use your REST API without a API key or to use it with a default key, which can be set programmable. Otherwise I would have to look for a different solution to control my devices.

All the best

Framspott

manup, 2018-02-28:
The API key is required and in fact future versions will even raise the bar on security matters.

**Note** Without an API key pretty much every website on the internet could fully access the API from the browser via javascript.

When using the HTTP basic authentication you can create an API key by using the password. The gateway doesn't need to be unlocked neither the WebApp needs to be used.

http://dresden-elektronik.github.io/deconz-rest-doc/authorization/

Kane610, 2018-02-28:
@manup would it be possible to get some draft (or documentation) of those improved security improvements so I can prepare for support?

manup, 2018-02-28:
Yes when it's time it will be developed and discussed here on GH. Currently and since a long time I'm looking into various techniques but it's very hard to find one which doesn't expect the users to be security experts. Most popular web enabled security mechanisms expect that the server (deCONZ) needs to be reachable over the internet, which is a pity. HTTPS and Letsencrypt certificates would be my favorite but they require the server to have a valid domain. ... So the Research goes on.

Kane610, 2018-02-28:
@manup I sent a message to a friend who is security architect at my company. Lets see if he can be available for a dialogue.

ebaauw, 2018-02-28:
Would it be possible to have the deCONZ web server check whether the API request comes from the loopback interface?  We wouldn't need authentication nor SSL when the deCONZ client runs on the same server, as will probably be the case for homebridge-hue and other home automation integrations.

In fact, security-wise, I would be happy with an option to run the deCONZ REST API only on the loopback interface (this might actually be a side-effect of #444, if we could specify 127.0.0.1).

thorne44, 2018-03-10:
Have to agree with ebaauw. I'm running HASS IO on a raspberry Pi with integrated Zigbee and deConz.

The scripts I'm using exist on the same server. It would be nice to not need the security for the same server.

manup, 2018-03-11:
In regard to the request to allow API usage without an apikey.

Consider a script from a malicious or tampered website which is opened in a browser *on the* Raspberry Pi. From the API point of view this *is a* local access from 127.0.0.1, the API can't reliably differentiate where the script comes from.

I understand the problem for easier setup but we need to find a better approach, maybe provide a apikey via commandline parameter or config.ini file, that way the above issue can't be exploited.

Kane610, 2018-03-11:
I agree. The key scheme is a simple layer of security. We shouldn't remove it. But new security schemes could potentially be optional

ebaauw, 2018-03-11:
> Consider a script from a malicious or tampered website which is opened in a browser _on the_ Raspberry Pi.

Fair point.  I wouldn't run a web browser on a headless server, but I appreciate you don't want to rely on that.  Still, the API key is a joke, as long as it's exchanged in cleartext.

thorne44, 2018-03-12:
For my raspberry pi, I don't have a screen, keyboard or mouse thus makes it very difficult to open a browser.

Why not make it an option to open up the security for local scripts? If it's off by default, people can have the choice to turn it on.

closed 2019-03-30 by stale[bot]