dresden-elektronik/deconz-rest-plugin #447: OSRAM Classic B40 firmware

roflmao, 2018-02-28:
Hi,

Im having some issues with connectivity with my OSRAM bulbs.
[image]

The 2etg_kjokken* bulbs which are my only OSRAM bulbs gets "grey" in Phoscon aka unreachable after some time.

How am i able to upgrade the fw on these with otau?
The current version is:
```
VendorOSRAM
ModelClassic B40 TW - LIGHTIFY
Version V1.03.20
```

ebaauw, 2018-02-28:
That's pretty old firmware indeed.  Afaik, the only way to update the OSRAM bulbs is through the Lightify gateway.  In theory deCONZ could do the firmware upgrade, but OSRAM haven't published their firmware files.

roflmao, 2018-02-28:
The bulbs worked great against my Hue hub. Any tips on getting firmware besides the OSRAM hub?

jonathanweinberg, 2018-03-12:
As a heads up, I’ve been able to update Osram Sylvania BR30 RGBW bulbs via a SmartThings Gen2 hub.

closed 2019-03-30 by stale[bot]