dresden-elektronik/deconz-rest-plugin #448: [Information] A new version of Xiaomi Aqara wireless switch

oleg77s, 2018-03-01:
[A new version of Xiaomi Aqara wireless switch](https://aliexpress.com/item/Original-Xiaomi-Aqara-Wireless-Switch-Zigbee-System-Remote-Control-Switch-Use-With-Xiaomi-Gateway-Aqara-Air/32826677539.html): clicking, double clicking, long pressing, shaking.
built-in high precision acceleration sensor

ebaauw, 2018-03-01:
Interesting!  On the ouside it looks like my Xiaomi Aqara smart switch, but I don’t think mine supports shaking.  Never tried, actually.  And I couldn’t get any long press, only single, double, treble, and quadruple presses.  Only one message per press/release - no separate press vs release messages.

johga1, 2018-03-02:
I have three of these. They only report buttonevent 7000 when performing a double press. 

Anything else is not visible in my logs.

I’m running Deconz GW 2.04.99 as a HASS.io 0.64.2 add-on. All in one headless pi3-box.

jurriaan, 2018-03-03:
I have two of these: https://www.aliexpress.com/item/Xiaomi-Aqara-Smart-Wireless-Switch-Key-Built-In-Gyro-Function-ZigBee-Wifi-Work-With-xiaomi/32847092257.html 

The description says 'Built-in gyro'. This made me think this switch should support that as well?

ebaauw, 2018-03-03:
Mine has the same type as the one on the aliexpress picture: WXKG11LM.  They could of course have used the wrong picture.
```
  "modelid": "lumi.sensor_switch.aq3",
```
Mine has `lumi.sensor_switch.aq2`.

> They only report buttonevent 7000 when performing a double press.

Afaik, the Xiaomi cube is the only device that previously could issue this buttonevent value.  It correspond to value 0x02 for the _Measured Value_ attribute (0x0055) of the _Multistate Input_ cluster (0x0012).  I'm guessing the new switch exposes this cluster instead of _OnOff_?  Could some-one please post a screenshot of the node in deCONZ, showing the clusters?

My guess is that the switch reports all actions through the _Multistate Input_ cluster, hence you don't see any other buttonevent values.  deCONZ should already update the _Measured Value_ in the GUI.  If some-one could post the values corresponding to the various actions, adding the corresponding buttonevents to deCONZ will be straightforward.  Here's a screenshot of the Cube:

[image]

jurriaan, 2018-03-03:
@ebaauw Mine is also a `lumi.sensor_switch.aq2`, and has no multistate input cluster unfortunately.

johga1, 2018-03-03:
Hope this helps.

Single:
[image]

Double:
[image]

Holding (while holding):
[image]

Hold (after release):
[image]

Shake:
[image]

ebaauw, 2018-03-03:
Thanks @johga1, that's just what the doctor ordered.  Created PR #455 which should do the trick.  I hope @manup can still merge it into 2.05.09 later today, or we'll have to wait for 2.05.10.

johga1, 2018-03-12:
Anyone else having trouble joining these?

I've been able to join my three Xiaomi Wireless Buttons but are now unable to do so. It worked on version 2.04.99 but then the buttons did not work. Now the buttons work on my one device that is joined to the network. 

My other two remotes that I'm trying to join does not show up in Rest.

I am able to join and see them through the deCONZ GUI/App, this does however not add them to Rest API and therefore unusable for me with Home Assistant. 

Anyone else seeing this problem, or is it something with my local config.

tuxinger, 2018-03-17:
Same here, I cannot use the WXKG12LM as they are not exposed by the api. As I run deconz headless I cannot check if it´s actually joined.

johga1, 2018-03-17:
It seems like there's been a bug introduced in 2.05.10 when they added support for the AQ3 devices (registering buttonevents which it didnt do earlier). 

I've been able to add my AQ3 devices on a 2.04.99 installation on my computer. I then did a backup and moved my configuration to my headless HASS.io with Deconz. Buttons now work, but I forgot to add my door sensor which also is AQ3. I'm having the same issue adding that one on 2.05.12.

ebaauw, 2018-03-17:
Haven’t seen an aq3 door sensor before.  What’s the model id for that one?  For all Xiaomi battery-powered devices: you want to be pressing/releasing the reset button (pin hole) like every second during pairing, to make sure it remains awake.  Also, the device should report its model id on pressing/releasing reset.  That, with the mac address and manufacturer code, should be enough for the REST API to recognise the device.

bdickhaus, 2018-03-19:
I was having the same problem here with a lumi.sensor_switch.aq3. Version 2.05.15. I was able to add it using version 2.04.99, and once added it works properly with 2.05.15.

andenmad, 2018-03-29:
hi 
i try to add wxkg12lm lumi.sensor_switch.aq3 switch. 

but its not seams to reginize the switch se picthure for details 

 
[image]

[image]

version running 
[image]

anyone know what im doing wrong

manup, 2018-03-29:
The Phoscon App won't display the switch yet.
Did you join it while sensor search was active?

*Phoscon App > Menu > Devices > Sensors > Add new sensor > Other*

Then reset the switch, not sure how it is done for this one but the other Xiaomi one button switch had a small button which had to be pressed a few seconds.

The switch should then be visible in the REST API.

andenmad, 2018-04-04:
yes i discover from the sensor search it dosent find a sensore it just appere with 0x0000 in deconz see picture aborve

andenmad, 2018-04-09:
i have just check and it dosent appere in REST API. 

i have upgraded to software version 2.05.20 and try to add sonsor again still sam result.

andenmad, 2018-04-10:
i was thinking it maybe was the first device i got there was a issue with but i just recieved a new one and it is same experience. 

is there other there experience the same ore anyone know why.

ebaauw, 2018-04-10:
I did receive some new Aqara door sensors earlier this week and during pairing I noticed that the _OnOff_ cluster only becomes visible in the deCONZ GUI when the magnet is moved (i.e. when the sensor sends its first attribute report).  The REST API resource is only created _after_ the _OnOff_ cluster is visible.  Just spamming the reset button is _not_ enough to get the deCONZ REST API to recognise the new sensors.

I suspect the same might be the case for the switch: during pairing you need to press the switch button during pairing (and maybe shake it as well), in addition to pressing the reset button.  I'm expecting delivery of the .aq3 switches this week, to be continued...

andenmad, 2018-04-10:
the sensor search keep standing like this 
[image]

and device is pressent in deconz like this. (have pres any key combination i could think of) 

[image]

[image]

single click 
[image]

double click
[image]

sharke 
[image]

hold 
[image]

release Hold
[image]

ebaauw, 2018-04-10:
Does the switch show as a `/sensors` resource in the REST API? The switch is not yet supported in the Phoscon app, as @manup mentioned earlier.

andenmad, 2018-04-11:
no it dosent show up in rest Api.

ebaauw, 2018-04-13:
Found that sensor was not whitelisted when processing _Multistate Input_ cluster.  Could some-one try my latest PR?

andenmad, 2018-04-16:
how do i implement PR. is it just change files ore do i need to build pakage ?

ebaauw, 2018-04-16:
You need to build the REST API plugin following instructions in the README.  Or wait for it to be included in the next release.

tbergler, 2018-06-12:
I have two of "lumi.sensor_switch.aq2" showing up in deCONZ. How can I actually use them to switch any light?

Jasonthefirst, 2018-09-21:
Is this problem fixed? Because I can't find the switch wxkg11lm in deconz but not on the web and therefore not in home assistant. Could you point me to the instructions to build the REST API?

manup, 2019-01-17:
Closing this for now, the switches should be supported since 2.05.50.

closed 2019-01-17 by manup