dresden-elektronik/deconz-rest-plugin #449: https://dresden-light.appspot.com/discover reports only one DeCONZ

androme13, 2018-03-02:
Hi i have two DeCONZ, only one is returned here : https://dresden-light.appspot.com/discover.
I think json string return only the first entry.

But with this URL, both are seen : http://dresden-elektronik.de/pwa/login.html#

manup, 2018-03-02:
The discover service will return all gateways which have a) discovery enabled and b) can reach out.
The pwa is a bit different since it also searches local gateways (in LAN) which aren't listet in the internet discovery service.

androme13, 2018-03-02:
i'm running my second deconz in headless mode, may be it's specific ?
Sorry for my lacks in english, i'm french.

manup, 2018-03-02:
All good :) headless mode shouldn't be a problem, since beside the hidden deCONZ GUI both versions are identical.

You may check in the old WebApp that `Internet Service` is enabled in:
 *Menu > Settings > Show Advanced Settings*:

androme13, 2018-03-02:
I already checked that, on the second deconz there is not deconz hardware (usb or gpio) yet, only on the first one, may be this is the cause ? but i can reach the web interface and the websocket port.

manup, 2018-03-02:
Yes that's indeed the cause, the discovery service only lists gateways with a RaspBee/ConBee hardware mac address.

androme13, 2018-03-02:
Thank you for your help, last question, is the way to achieve a scan like "The pwa is a bit different since it also searches local gateways (in LAN)" documented somewhere ?

manup, 2018-03-02:
It's not documented, but it works like this:

* detect own ip subnet(s) like 192.168.2.x via WebRTC/ trial n error
* query all hosts 192.168.2.1–254 on this subnet(s)

The query is: `GET http://192.168.2.1/api/config`

It's a bit brute-force but Javascript in browsers is very limited in capabilities to detect devices in LAN.

androme13, 2018-03-02:
Thanks a lot for your advices.

closed 2018-03-02 by androme13