dresden-elektronik/deconz-rest-plugin #450: Sometime an IKEA light goes dead, is there a way to get it back without power cycle ?

donnib, 2018-03-02:
Hi @manup 
I have seen this few times that some IKEA bulbs does not react anymore and i need to do a power cycle before they work again. I have tried restart deconz and also `sudo GCFFlasher_internal -r` and also type 0 while the light is selected. Nothing i do can get i back. Doing a power cycle is a hassle because they are on always power so it's not easy to do this. The only thing i have not done is a complete power cycle of the rPi.

/donnib

manup, 2018-03-02:
Hi @donnib,
If the other lights are still working it might be the light itself which gets stuck, in that case restarting deCONZ/RaspBee or the gateway won't have an effect.

The latest firmware 0x261E0500 and deCONZ 2.05.08 contain some IKEA related firmware improvements, which version are you running?

donnib, 2018-03-02:
@manup yes the other lights were working ok but just this one had problems. I am running 0x261E0500 and deCONZ 2.05.05. I will upgrade to 2.05.08. If this fixes anything i presume this might make this situation not repeat itself rather than get the light back again ?

manup, 2018-03-02:
Hmm if you're already running 0x261E0500 I don't know what may causing the problem :/ 
Do the IKEA lights all have the recent firmware?

tleegaard, 2018-03-04:
In my experience the IKEA bulbs get unresponsive while transitioning. If I do a 5 minute transition and try to send a new command after 1 minute the bulb will not respond. Waiting the whole 5 minutes the bulb becomes responsive again. Same issue proposed here: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/384#issuecomment-362248078

closed 2019-03-30 by stale[bot]