dresden-elektronik/deconz-rest-plugin #452: Phoscon didn't find the gateway

joecas1, 2018-03-03:
Hi all,

just want to try out the new Phoscon App. If I open "www.dresden-elektronik.de/pwa", the app started and after a few seconds the ip-address of my gateway is showing up in the head line. The circle is moving for ~2-3 minutes and afterwards it shows a blank page with just the Phoscon logo ... nothing else. :(
How can I get it working?

Joe

manup, 2018-03-03:
If you know the ip address of the raspberry (or PC) you can open the Phoscon App locally:

    http://<ip-address>/pwa

If deCONZ is running on a other port than 80 please use:

    http://<ip-address>:<port>/pwa

**Note** It's recommend to use the latest available version (currently 2.05.08).

https://www.dresden-elektronik.de/rpi/deconz/beta/

joecas1, 2018-03-03:
PERFECT! Something went wrong during upgrade to 2.05.08 ... just repeated everything and now it's working! Thanks for the reminder ... :)

closed 2019-03-30 by stale[bot]