dresden-elektronik/deconz-rest-plugin #453: after update 2.5.8 works as accesspoint

s1738berger, 2018-03-03:
Hi,after the update to th latest beta 2.5.8 the rest api work as a wifitype =  'accesspoint'.
Where is the easiest way to configure it back to connect to an existing network as configured in '/etc/wpa_supplicant/wpa_supplicant.conf'?
Thanks for your help
regards Stefan

balu-, 2018-03-03:
soo the deconz-wifi.service is only needed in your sd-card images?
(a already disabled it because my pi does not have wifi hardware)

s1738berger, 2018-03-05:
Thank for the quick solution, manup.
It works, in the file **/etc/dhcpcd.conf** I had to uncommend at the end:
```
# added by deCONZ
# interface wlan0
# 	static ip_address=192.168.8.1/24

```
Btw. the latest version 2.5.10 still has the same behaviour. The steps I do are:

- Install image: Phoscon_Gatway_2017-12-10.img
That will bring the latest stable version, working as client
 - Update to 2.5.10 with:
sudo dpkg -i deconz-2.05.10-qt5.deb, , now works as accesspoint