dresden-elektronik/deconz-rest-plugin #454: Hue Tap is showed greyed out

joecas1, 2018-03-03:
In the group menu the Hue Tap is showed greyed out (disabled) and it's not possible to set up any new scene to a button. Did I miss anything or is the Hue Tap functionality not implemented completely right now?!

deimos84sk, 2019-03-23:
Hi,
I got the same issue on my side. I'm trying to connect my hue tap but cannot select it in the add menu. How can I connect it nowadays, since regarding to the supported device page, hue tap was or still is supported.

thanks!

deimos84sk, 2019-03-25:
Ok, the issue is solved:

When logging in via _http://www.dresden-elektronik.de/pwa/index.htm_ Hue Tap is greyed out. I had this situation since the first installation of phoscon.

Now, if you log in via you local IP (_http://<the IP of your gateway>:8080/pwa/index.html_) however, you can add  a Hue Tap device and configure it easily.

Tested with:

Hersteller dresden elektronik
Produkt ConBee
Version 2.05.60 / 10.3.2019
Firmware 26320500

closed 2019-07-30 by stale[bot]