dresden-elektronik/deconz-rest-plugin #456: Ikea wireless dimmer

Kane610, 2018-03-03:
What is the current level of support for the Ikea wireless dimmer?

I noticed that Phoscon has it greyed out and when I added lights to the group of the dimmer only a couple of those lights where controlled by it.

/R

hellms, 2018-03-04:
moin !
will piggyback my question to this topic.
my ikea wireless dimmer is runnig firmware 1.2.248. (updated with deconz and the ikea firmare file, works like a charm ;-)
ah, in the meantime i´m using deconz 2.05.10.
the dimmer ist recognized as an IKEA tradfri wireless dimmer with a nice circle icon, guess that came with the 2.0.5.x beta.
but no dimming fuction at all to any of the lights (checkend ikea and osram bulbs)
connected directly to ikea bulbs works, so the dimmer itself is ok.
in deconz with "signal monitor"  i can see counter rising when i´m turning the dimmer.

so any hints what i should check next ?
cheers

mploetne, 2018-03-06:
Hi,
i have noticed the same behavior.
when i connect a new ikea dimmer thru the pwa wizard, the dimmer shows up after doing the 4 press reset and i'm promted to turn it clockwise. but that move is not recognized and i run in a timeout.
the group is created but its greyed out in the pwa. also the rules are created.

Marko

mploetne, 2018-03-06:
sorry,
forgot to state im running running phoscon 2.05.10

ebaauw, 2018-03-07:
I have some funny behaviour of my IKEA Trådfri wireless dimmer as well.  Sometimes it would not register any buttonevents, until I re-pair the dimmer.

I've been playing around with the dimmer in deCONZ GUI while sniffing its traffic, and come to some funny conclusions:
- The dimmer does not seem to honour group bindings: it reports that the binding has been created successfully, but it won't send any group commands;
- The dimmer only sends _Level Control_ cluster (0x0008) commands, but it seems to ignore the cluster in binding table; even binding the _Basic_ cluster to a light makes the dimmer control that light;
- In particular, the binding deCONZ creates for the _Power Configuration_ cluster (0x0001) for _Battery Percentage Remaining_ attribute (0x0021) reporting, causes the dimmer to send unicast _Level Control_ commands to the RaspBee;
- The dimmer supports multiple bindings to control multiple lights simultaneously.  It sends different unicast commands, for each binding table entry.

I suspect that when deCONZ re-asserts the group binding (or maybe the battery binding), somehow the binding to the RaspBee gets ignored, removed, overwritten, otherwise invalidated, causing the dimmer commands no longer to send unicast commands to the RapsBee.  I don't suppose the RaspBee can eavesdrop on a unicast command from the dimmer to a light, hence no more buttonevent.

manup, 2018-03-07:
On reddit someone posted a url were IKEA test OTA firmware can be downloaded:

https://www.reddit.com/r/tradfri/comments/74qj5l/when_is_the_google_assistantalexahomekit_update/

Also this script, no changelog available though ...

```bash
#!/bin/bash

FEED_LOCATION="http://fw.ota.homesmart.ikea.net/feed/version_info.json"
FEED_LOCATION_TEST="http://fw.test.ota.homesmart.ikea.net/feed/version_info.json"

for FEED in "$FEED_LOCATION" "$FEED_LOCATION_TEST";
do
    wget --quiet -x -N "$FEED"

    for FIRMWARE_IMAGE in  $(curl -s -S "$FEED" | jq -r '.[].fw_binary_url');
    do
        wget --quiet -nc -x "$FIRMWARE_IMAGE"
    done
done
```

Kane610, 2018-03-08:
After some more fiddling with the dimmer it now doesn't do anything with any light and I don't get any updates in deconz either.

Gonna try to pair it again.

hellms, 2018-03-24:
short update:
updated to deconz-2.05.16 and flashed included firmware to my raspbee today.
to be on the safe side i deleted my (not working) tradfri dimmer from deconz.
freshly searched and connected it works like a charm now ;-)

many thanks to the hard-working developers !
cheers

Kane610, 2018-04-09:
After upgrading to 2.05.20 and upgrading to latest firmware I now got one of my two dimmers back in to the system. I added some lights but deconz says that it doesn't have a button handler for the dimmer. I can see signals go through over the web socket

Kane610, 2018-04-11:
By pushing the button once after adding lights to the dimmer group in Phoscon, the dimmer will pair to them. But only a subset of the (9) lights, different every time, gets added. Any tips here?

sjostrand, 2018-04-17:
Thanks @Kane610 clicking the button after adding light to the group did the trick.

But I'm experiencing the same problem with sync as you. The lights (3 in my case) are never at the same dimming level, they don't always turn off/on att the same time.

strips, 2018-10-23:
I have paired a Trådfri Dimmer several times now but unable to dim a Trådfri colour bulb. Other Trådfri remotes are able to dim the bulb.
Dimmer FW 1.2.248
deCONZ v.2.05.42

This is my first trial in deCONZ and using a ConBee. Conbee FW was upgraded today.

Gronsak, 2018-12-11:
So I'm running on Hass.io as a docker image meaning I only have access to Phoscon for settings.

I paired a dimmer a couple of days back and it made a group but I discovered that the firmware on the dimmer was old and didn't want to cooperate so I removed it, unpaired the dimmer and connected it to my trådfri gateway for updates and then back again.

Second time around I didn't get a group made for the dimmer and I can't see any way of manually making one or in any other way pairing the dimmer with lights in the webapp.

Am I missing something? Shouldn't a group be made like the first time I paired or does it remember the dimmer in the background?

rutberg, 2019-03-03:
I'm trying to work around this issue by binding the dimmer directly to the light, I assume the method is called touch link. But as soon as I do this, the light disappears from the gateway, showing as unavailable. To get it back again, I need to re-add it. This breaks the connection with the trådfri dimmer. Anyone who've gotten this to work?

dcshoecomp, 2019-03-21:
Dimmer FW 1.2.248
deCONZ 2.05.60

Can sometimes get it to pair with turn it clockwise, however when add lights to the switch they do nothing at all. This is still an issue with this dimmer.

cyclops1982, 2019-04-07:
HI All,

So, i have both the dimmer and the remote purchased recently.

They work, but the pairing process had an issue for me.
In the dialog in the deCONZ site, you are asked to press a button 5 times or turn the dimmer around. The 'next' button never seems to enable there, meaning that it doesn't seem to recognize it. If you cancel out of that, then suddenly the dimmer and remote do show up. You can then use them in the various light. It works fine.

deCONZ is version  260B0500 /  2.05.59 / 2/14/2019
TRADFRI remote control is version 1.2.214
TRADFRI wireless dimmer is version 1.2.248

One tip for the people with dimmer, it seems to be a auto-calibrated gyroscope in there. It was odd for me to see how it worked. I'm amazed by how small it is (great stuff!).
The gyroscope means that you basically have to stick it to a fixed location before using. The dimmer has two parts, but the base doesn't seem to have anything in it, so i believe some people are also struggling just because of how the dimmer works.

I personally use home assistant and the REST interface. In home assistant, they seem to just report battery status. I'd like to do automation based on the remote. Is there somebody who has that working?

Kane610, 2019-04-07:
Start by upgrading firmware to 262f or later

Slychocobo, 2019-04-07:
Seem to be having similar issues having inputs from the Dimmer being detected.

Running 
Ikea Tradfri Dimmer version  1.2.248
Conbee Version 2.05.59 / 2/14/2019
Firmware 262F0500

Can pair the dimmer via Phoscon (Device is detected & 'Rotate Clockwise' step can be completed) but after pairing it does not control any bulbs it is assigned to.

Tried removing/repairing the controller and lights a few times but having no success.

rutberg, 2019-04-07:
Same here, only difference is running FW  261F0500

cyclops1982, 2019-04-07:
@Kane610 Is there some magic way to do that? Software says "This version is up to date"

Kane610, 2019-04-07:
> @Kane610 Is there some magic way to do that? Software says "This version is up to date"

How are you running deconz? With the container the update script you should run identifies updates properly afaik

cyclops1982, 2019-04-08:
> 
> 
> > @Kane610 Is there some magic way to do that? Software says "This version is up to date"
> 
> How are you running deconz? With the container the update script you should run identifies updates properly afaik

Running deCONZ on ubuntu 16. Installed from DEB file using the rasbian steps. 
The user interface and version seems to be correct with the release on github. Is the version you refer to some beta version of specific build? If so - where do i get it from? Or maybe better, how do i build deCONZ myself?

ebaauw, 2019-04-08:
> how do i build deCONZ myself?

You don't.  The only part that's open source, and you can build yourself, is the REST API plugin.  Even for that, building is only supported on Raspbian.  See README.

cyclops1982, 2019-04-09:
> 
> 
> > how do i build deCONZ myself?
> 
> You don't. The only part that's open source, and you can build yourself, is the REST API plugin. Even for that, building is only supported on Raspbian. See README.

Thanks!

cyclops1982, 2019-04-09:
So, although the dimmer works, i can't (in the website) change to which lamp it is assigned to. I've tried to do this and the UI indicates that it has changed, but it doesn't work.

Where do i report these issues? How can i troubleshoot these further ?

MickL, 2019-06-28:
Hey guys, how do you know which firmware the dimmer has?

I have a Tradfri dimmer and a remote. With both I struggled connecting them. It says i have to press the button 4 times but the next button does not activates. For both devices it did not work for 10-20 times i tried, and then suddenly when i am about to give up it works?! Is there a problem with the devices, phoscon/deconz or the help text? Maybe 4 times pressing is not correct? It says it will blink after it is connected, but it blinks immidiatly even after only 1 press.

Anyway the remote is working fine (except for left/right to change temperature but thats another issue ) but the dimmer does not change anything attached to its group...

timwegenerDE, 2019-10-24:
I was able to add the dimmer to phoscon. The trick for pairing: Wait a few secons - turn it a bit to the right - Wait a few seconds and don't move the dimmer while waiting (not the wheel and not the dimmer itself). Otherwise try it again.

Now the dimmer is connected to phoscon but doesn't do anything even when a light is associated......

Was anyone able to get it working?

timwegenerDE, 2019-10-24:
Is it possible that it doesn't work with bulbs that have color spectrum?

furious1y, 2019-11-19:
> I was able to add the dimmer to phoscon. The trick for pairing: Wait a few secons - turn it a bit to the right - Wait a few seconds and don't move the dimmer while waiting (not the wheel and not the [...]

Some other kind user posted on another thread here.

There's a gryo inside the sensor - if you dont have it perfectly upright when adding, it won't register the turn clockwise, and that won't complete the addition to the network.

In short - stick it to a wall. take the cover off and press 4 times to start. Once it goes to "turn clockwise shortly", let the magnet grab and stick it to the wall and twist it clockwise. If you've got it right, it'll instantly update and you now have a working sensor......

Talk about finicky....

timwegenerDE, 2019-11-19:
> Some other kind user posted on another thread here.
>
> There's a gryo inside the sensor - if you dont have it perfectly upright when adding, it won't register the turn clockwise, and that won't [...]

I also could add it into phoscon, but I can‘t control any lights with it. the connected lights just not changing the brightness. Maybe because they are all color-changing bulbs and phoscon don‘t know how to handle them? 

You could control your lights?

henkkakay, 2019-11-19:
Interestingly I managed to pair/connect this - added it to my OpenHab installation too. And when I turn it, OpenHab registers buttonevent and button change. But the values are seemingly random (switches between 2002 and 3002 mostly, but occasionally goes to 1002 and 4002 as well) - irrespective of whether I am turning it clockwise or anti-clockwise, in quick+small turns or bigger turns. So in practice I cant use it for anything :-( 

An extract of events.log below

`2019-11-20 00:04:23.324 [vent.ItemStateChangedEvent] - deconz_switch_f9384011_000b57fffe906b4e011000_button changed from 2002 to 3002
2019-11-20 00:04:23.329 [vent.ItemStateChangedEvent] - deconz_switch_f9384011_000b57fffe906b4e011000_last_updated changed from 2019-11-20T00:04:22.000+0200 to 2019-11-20T00:04:23.000+0200
2019-11-20 00:04:23.686 [vent.ChannelTriggeredEvent] - deconz:switch:f9384011:000b57fffe906b4e011000:buttonevent triggered 2002
2019-11-20 00:04:23.697 [vent.ItemStateChangedEvent] - deconz_switch_f9384011_000b57fffe906b4e011000_button changed from 3002 to 2002
2019-11-20 00:04:24.302 [vent.ChannelTriggeredEvent] - deconz:switch:f9384011:000b57fffe906b4e011000:buttonevent triggered 3002
2019-11-20 00:04:24.316 [vent.ItemStateChangedEvent] - deconz_switch_f9384011_000b57fffe906b4e011000_button changed from 2002 to 3002
2019-11-20 00:04:24.322 [vent.ItemStateChangedEvent] - deconz_switch_f9384011_000b57fffe906b4e011000_last_updated changed from 2019-11-20T00:04:23.000+0200 to 2019-11-20T00:04:24.000+0200
2019-11-20 00:04:24.630 [vent.ChannelTriggeredEvent] - deconz:switch:f9384011:000b57fffe906b4e011000:buttonevent triggered 2002
2019-11-20 00:04:24.645 [vent.ItemStateChangedEvent] - deconz_switch_f9384011_000b57fffe906b4e011000_button changed from 3002 to 2002
2019-11-20 00:04:24.955 [vent.ChannelTriggeredEvent] - deconz:switch:f9384011:000b57fffe906b4e011000:buttonevent triggered 3002
2019-11-20 00:04:24.969 [vent.ItemStateChangedEvent] - deconz_switch_f9384011_000b57fffe906b4e011000_button changed from 2002 to 3002
2019-11-20 00:04:25.267 [vent.ChannelTriggeredEvent] - deconz:switch:f9384011:000b57fffe906b4e011000:buttonevent triggered 2002
2019-11-20 00:04:25.300 [vent.ItemStateChangedEvent] - deconz_switch_f9384011_000b57fffe906b4e011000_button changed from 3002 to 2002
2019-11-20 00:04:25.309 [vent.ItemStateChangedEvent] - deconz_switch_f9384011_000b57fffe906b4e011000_last_updated changed from 2019-11-20T00:04:24.000+0200 to 2019-11-20T00:04:25.000+0200
2019-11-20 00:04:25.569 [vent.ChannelTriggeredEvent] - deconz:switch:f9384011:000b57fffe906b4e011000:buttonevent triggered 1002
2019-11-20 00:04:25.581 [vent.ItemStateChangedEvent] - deconz_switch_f9384011_000b57fffe906b4e011000_button changed from 2002 to 1002
`

wwebers, 2019-12-15:
@henkkakay 
I too bound this remote to OpenHAB (via deconz) and found the following: The remote reports 4 button values:

1002: fast turn clockwise
2002: normal turn clockwise
3002: normal turn counter-clockwise
4002: fast turn counter-clockwise 

Hope this helps. Should be enough to roughly write a rule dimming the light up or down depending on those values.

henkkakay, 2020-04-15:
A bit late reply sorry @wwebers - I hadnt checked the version - but is there easy way to do this software upgrade while the unit is paired/connected to DeConz and not Ikea app?

ebaauw, 2020-04-15:
https://github.com/dresden-elektronik/deconz-ota-plugin

benblb, 2020-05-19:
I'm able to get two of these dimmers connected in Deconz --> Node Red. Using the debug node I can see 4002, 3002, 2002... but not 1002. 

Over the course of a couple hours I haven't managed to get it to come up one time. During calibration I had one of the switches on the wall. 

Is this a known issue? Any way around it?

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix