dresden-elektronik/deconz-rest-plugin #457: Button triggers without button press

MrPsi, 2018-03-03:
I'm running Home Assistant and toggles a light with a Xiaomi Aqara Button using the deConz REST API. I installed version 2.05.09 today and now the light has started to turn on and off by it self, without anyone pressing the button. Has there been any changes that makes the REST API trigger a button event by it self?

MrPsi, 2018-03-04:
So is this an issue with the deConz REST API implementation in Home Assistant, or an issue with deConz?

manup, 2018-03-04:
> So is this an issue with the deConz REST API implementation in Home Assistant, or an issue with deConz?

Depends, I don't how Home Assistant triggers actions based on events.

However due changes in 2.05.10 the problem shouldn't occur anymore.

Kane610, 2018-03-04:
@MrPsi please enable debug log in hass to get more quantitative data. This might be something with report of updated states over web socket. Which have been a reason for "ghost" events earlier.

Regardless of the issue you can set constraints to your automations to make sure that you only act on changed states. Unless of course the reported state changes without the button actually being pushed.

/R

MrPsi, 2018-03-05:
The problem disappeared in version 2.05.10, so all seems good. Thanks @manup 

@Kane610 the button presses are events in Home Assistant, I don't think you have access to any previous state then?

Kane610, 2018-03-05:
@mrpsi yes but in the backend everything is a state. So for example if the battery gets updated there is an update to the buttons general state

MrPsi, 2018-03-05:
@Kane610 alright, thanks for the help. Will check it out if the problem arises again.

closed 2018-03-05 by MrPsi