dresden-elektronik/deconz-rest-plugin #459: Feature: Get light/sensor from uniqueid

jonatanolofsson, 2018-03-04:
The invariant of a zigbee device is its ieee id and endpoint id - in deconz combined as the uniqueid. It would be useful to be able to address the REST api using this instead of the assigned node id (which will change between repairings).

My suggestion is to replace getLightNodeForId with a getter that accepts either node id or, if the node id is not found, parses the string as a deCONZ::Address and endpoint and retrieves e.g. lights through getLightNodeForAddress

manup, 2018-03-04:
> My suggestion is to replace getLightNodeForId with a getter that accepts either node id or, if the node id is not found, parses the string as a deCONZ::Address and endpoint and retrieves e.g. lights through getLightNodeForAddress

Good point, it would improve lifetime of applications as well, since id might change for reasons like reset, deleting and rejoin devices or simply transfer to another ConBee/RaspBee setup/machine.

jonatanolofsson, 2018-03-04:
Should the uniqueid be sent on the websocket as well?

jonatanolofsson, 2018-03-04:
Also, the equivalent would of course also apply to sensors

closed 2019-03-30 by stale[bot]