dresden-elektronik/deconz-rest-plugin #460: v2.05.09 GUI changes

ebaauw, 2018-03-04:
Love the new GUI on v2.05.09 on Raspberry Stretch with the curved lines.  Also love that the node box stretches to show the full name.  Unfortunately, the mac address is now a but smaller, and I can no longer read it.  Granted, this is an issue of my eyes more than a deCONZ issue, but would it be possible to increase the font size for the mac address?

[image]

manup, 2018-03-04:
:) cool, yes I was so annoyed not seeing the full names that I finally fixed this, was on the todo list for a long time. The curved lines were disabled in 2013 on Raspberry Pi since the old RPi 1 was too slow, with RPi 2/3 this shouldn't be an issue any longer.

Also the two buttons on the right of the node should now be more legit due more contrast.

Yes I'll adjust the mac address font size to some points larger in the next version, there is enough space for this. On PC this already looks right, RPi seems to have some different font metrics..

wvuyk, 2018-03-04:
Manup, thanks for this improvement! I finally am no longer gambing where these buttons are on the node.
Love this one 👍

manup, 2018-03-04:
Glad to hear, in 2.05.10 hopefully the mac address should be more readable too.

ebaauw, 2018-03-04:
Yes it is!  Thanks.

closed 2018-03-04 by ebaauw