dresden-elektronik/deconz-rest-plugin #461: PWA and Home Assistant won't work with new version

FezVrasta, 2018-03-04:
I'm not sure if it's just because of the new version (2.05.10-qt5) or because I added the `--http-listen` option to the start command.

Anyway now both the official PWA and the Home Assistant won't discover DeCONZ.
Also, if I define the IP manually in Home Assistant, it won't be able to connect.

FezVrasta, 2018-03-04:
I tried without the `--http-listen` and it still doesn't work ("can't get API key"). So it seems like it's the version the problem

manup, 2018-03-04:
Can you provide a error log from Home Assistant?
Do you run deCONZ on a other port than 80 or 8080?

FezVrasta, 2018-03-04:
I ran it on port 80. I'll try to provide a log ASAP

marthoc, 2018-03-06:
@FezVrasta you may also want to raise an Issue for @Kane610 to look at the Home Assistant side: https://github.com/kane610/deconz.

FezVrasta, 2018-03-06:
I doubt it's an home-assistant problem because it can be reproduced with the official Dresden PWA

Kane610, 2018-03-06:
I'm awaiting the logs :)

FezVrasta, 2018-03-06:
I doubt I'll be able to provide them anytime soon unfortunately, my home is the "production" one and I can't afford to stay without lights :-( wifey would get mad

Kane610, 2018-03-06:
Enable debug with logger component, upgrade deconz, restart hass, collect logs, fall back to old version, profit 👍 15 minutes at most, doesn't really affect the lights

FezVrasta, 2018-03-12:
Umh, I was going to do some test and I noticed I'm on 2.05.10 even if I remember I did restore the old 2.04.x version... I guess I accidentally installed again the same deb package and that fixed the problem I was having with the PWA app and Home Assistant.

closed 2018-03-12 by FezVrasta