dresden-elektronik/deconz-rest-plugin #463: Busch-Jaeger / Wall mounted transmitter 2-channel :: Can only assign scenes to one pair of buttons

erikproper, 2018-03-05:
Guess the subject says most of it.
Have a Busch-Jaeger, wall mounted transmitter 2-channel. So,
that is the battery powered one. 
Firmware version 1.2.0 is.
In Phoscan, I can only assign scenes to the lower pair switches. 
The upper one only lets me switch on / off the assigned lights. 
Note: This is the battery powered one ...

manup, 2018-03-06:
Yes that's a fixed setup, chosen to match the JUNG and GIRA switches which have this fixed setup by design.

We are currently working on a new switch editor in the Phoscon App where buttons can be assigned to a custom actions like calling a scene. But it will take a while till it's available.

erikproper, 2018-03-06:
Okay! Good to know that there's nothing wrong with the switch then ;-)
Will await future updates ;-)

closed 2018-03-06 by erikproper

Daniel0984, 2018-03-06:
This is same same fault i mentioned here: 368

manup, 2018-03-06:
Hey guys for your interest here is a graphic mockup with a Busch-Jaeger 4-channel switch which shows how we imagine the editor (German).

[Tastenbelegung.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/1786097/Tastenbelegung.pdf)

Basically various actions can be assigned to a switch button. Also each button may address another target group.

Therefore for each button you can 
* select a target group
* select the action

There will be also some special actions like *activate next scene* to cycle through scenes of a group with just one button.

This should also address the issues mentioned in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/368

Daniel0984, 2018-03-06:
That would be awesome! :thumbsup:

erikproper, 2018-03-06:
Looks good!

Wyphorn, 2018-10-07:
> Hey guys for your interest here is a graphic mockup with a Busch-Jaeger 4-channel switch which shows how we imagine the editor (German). [...]

Is there a release date for this?