dresden-elektronik/deconz-rest-plugin #464: ELKO dimmer latency

okseri, 2018-03-06:
I am trying to add a ELKO-dimmer to deConz (using ConBee) an I am experiencing some latency problems that I hope you can help with. 

Dimmer model:	ELKO ESH Plus 316GLED dimmer RF http://bit.ly/2oLyvxs
deConz version:	2.04.99

As the dimmer is not officially supported I cannot add it through Phoscon, but when searching for devices in the WebApp, it is found and adds to my environment. It does, however, add itself as a lightbulb (but I don’t think this matters?)

My main problem after adding the dimmer is that any physical changes on the dimmer take approximately 15-20 seconds before they register in the deConz interface. If I change settings within the deConz application (ON/OFF) the dimmer responds immediately.

From what I have read so far I guess this has something to do with the dimmer not being configured by the Phoscon and therefore not “communicating directly”, but instead sending Report Attributes commands (probably every 15 seconds or so).

I would be super thankful if you could give me any feedback on this.
**_-	Is it possible to have this dimmer supported/integrated, or is there anything I can do to reduce signal latency?_**

[image]

ebaauw, 2018-03-06:
> It does, however, add itself as a lightbulb (but I don’t think this matters?)

That’s because, from a ZigBee perspective, it _is_ a light.

> My main problem after adding the dimmer is that any physical changes on the dimmer take approximately 15-20 seconds before they register in the deConz interface.

That would be the delay in deCONZ polling the dimmer for its current state.

> From what I have read so far I guess this has something to do with the dimmer not being configured by the Phoscon and therefore not “communicating directly”, but instead sending Report Attributes commands (probably every 15 seconds or so).

Close.  The setup of attribute reporting is done by the deCONZ REST API plugin itself, not by the Phoscon app.  If the dimmer supports attribute reporting, deCONZ should configure it so that it’ll send a report when the state changes.

It might take some time for deCONZ to do this setup, however.  You might want to try and configure this manually.

ebaauw, 2018-03-06:
Have a look at the screenshots here: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/340#issuecomment-353648984.  You need to configure reporting on the 0x0000 attribute of the _OnOff_ cluster and of the _Level Control_ cluster.  And you need to create a binding from each of these clusters to the ConBee.

okseri, 2018-03-07:
That worked like a charm! Thank you, @ebaauw!

I set the Report Interval to Min 1, Max 300 and Reportable Change 1 and the dimmer response is acceptable.

Two more questions if you don't mind:

- Do you have any recommendations for a dimmer switch in a network with 5 dimmers and approx 30 HUE lights?
- I create the cluster bindings and get Success, but I cannot see the bindings in deCONZ. Is it possible to visually see the bindings in the GUI?

ebaauw, 2018-03-07:
> Do you have any recommendations on reporting interval settings in a network with 5 dimmers and approx 30 HUE lights?

I don't think the network size is a consideration.  The main concern would be battery consumption of the sensor, but the dimmer is mains powered.  As for the settings: use the Source, Luke:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/2f7fdf651ba59a73c50617f851083d546e48e933/bindings.cpp#L867-L883 and https://github.com/dresden-elektronik/deconz-rest-plugin/blob/2f7fdf651ba59a73c50617f851083d546e48e933/bindings.cpp#L916-L934

> I create the cluster bindings and get Success, but I cannot see the bindings in deCONZ. Is it possible to visually see the bindings in the GUI?

No.  The only "workaround" I've found: unbind the cluster, if that reports success, there was a binding, otherwise it reports not found.

closed 2018-03-07 by okseri