dresden-elektronik/deconz-rest-plugin #465: Feature Request: configurable user data directory

marthoc, 2018-03-06:
@manup - can you allow deCONZ to use an arbitrary directory for storing user data? Currently it seems like different aspects of user data get stored in different places on the Linux filesystem, which is making it hard in Docker development to “persist” a user’s configuration (when a Docker container is deleted and recreated, such as during a version update, all aspects of the filesystem get recreated except for what has been persisted). For example, I’ve tried persisting $HOME/.local/share/dresden-elektronik/deCONZ, but there must be more data than this as on starting the new container no groups, names, etc are present. 

Any ideas on this?

manup, 2018-03-08:
@marthoc deCONZ all configuration data used by deCONZ is indeed stored in `$HOME/.local/share/dresden-elektronik/deCONZ`. If needed we can add an command line option to use a different directory, this might also become handy in debugging situations.

Beside that only ZigBee network settings &mdash; like panid, network key, channel, etc. &mdash; are stored on the RaspBee/ConBee directly. Future versions will keep these parameters also in the configuration directory, for more reliable network operation.

marthoc, 2018-03-08:
Thanks @manup. I think my particular problem is related to developing for Hass.io (the containerized version of Home Assistant) - but a command-line option to store user data would be very helpful in this situation and the general Docker scenario, and you are right probably for debugging purposes too.

snozzlebert, 2018-03-09:
@marthoc You can use a symbolic link or for the Hass.io addon you can change the location of the home directory
```
export HOME=/data
```
or at Docker level:
```
ENV HOME=/data
```

marthoc, 2018-03-09:
@snozzlebert Yes, changing the homedir is what I’m doing now in my Hass.io addon. Unfortunately it’s not especially transparent to the user where the data is - even if you change the homedir to, for example, /config/deconz, it means that the data gets written to /config/deconz/.local/share/dresden-elektronik/deCONZ. Which means for the average user /config/deconz looks empty because most clients hide dot files by default. Would be much better to be able to specify the data dir as /config/deconz and have the files appear in there directly. I’m not crazy about a symlink because its a workaround and probably more prone to breaking.

closed 2019-03-30 by stale[bot]