dresden-elektronik/deconz-rest-plugin #466: Add support for Ikea Tradfri LED drivers

RudolfRendier, 2018-03-09:
Ikea is adding led drivers to its Tradfri line of products.
(Two models: max 10W and max 30W)

https://www.iculture.nl/nieuws/tradfri-drivers-bestaande-ikea-verlichting/

They can be controlled with the puck-dimmer (and paired to the hub)
It seems that they are already available in Danish and Dutch stores.

I'm planning on buying these; how can I help to get them supported in Deconz?

manup, 2018-03-09:
Since these are lights they should already work out of the box, special handling is mostly needed for sensors and switches.

RudolfRendier, 2018-03-10:
I can't get them to join the network.

I normally run deCONZ headless on a pi, but I can stop the deCONZ service and run it manually with X11 Forwarding, but I don't see a new device in the network.

Tried running with --dbg-info=2 and the only thing that stood out was:
'13:19:09:527 Unsupported device protocol version: 0x0106'
But I doubt it's related to these devices; there's a lot going on right after starting deCONZ. :)

Not sure what commandline arguments provide useful information for pairing new devices.

manup, 2018-03-10:
The you may need to reset the device with the *IKEA reset sequence* while light search in the Phoscon App is active.

https://www.youtube.com/watch?v=mJm9YpPrGzk

dxxxm, 2018-03-13:
Available at (some?) IKEA stores in Berlin. Reset method different for the new ones. They got a dedicated 'Reset' push button. To reset use a pin and push button for about 5 sec.
Didn't had any problem to hook them up to deCONZ, e.g. PHOSCON reading:
```
model TRADFRI transformer 30W
Version 1.2.245
```

RudolfRendier, 2018-03-27:
Can confirm it works as advertised.

Manual doesn't mention holding the button for 5 seconds, but that certainly seems to do the trick.
I was messing with the wrong cables before, but yesterday I was able to pair both types (10W and 30W)

It seems as though the devices accept just one reset per power cycle. So if the pairing fails, make sure to cycle the power once before your push the pin again.

closed 2018-03-27 by RudolfRendier