dresden-elektronik/deconz-rest-plugin #470: Mesh network not working with Tradfri remotes

FezVrasta, 2018-03-12:
I have 3 IKEA Tradfri remotes paired with my RaspBee, as far I can see they don't create a mesh network as I would have expected.

I have a remote that is too far from the RaspBee and most of the times it doesn't respond. I tried to place a remote at middle distance between the RaspBee and the problematic remote and nothing changed.

I also tried with the other remote to make sure it wasn't a problem of the remote itself and the result is the same.

Is there any way to diagnose what's happening?

manup, 2018-03-12:
Remotes can't create a mesh network since they are battery powered devices, they sleep most of the time (radio is off). To improve your network you need to put mains powered devices in the related areas. For example ZigBee lights and power plugs build a mesh to which the remotes (and sensors) can connect to.

FezVrasta, 2018-03-12:
Oh that makes sense! Thanks for the quick reply.

closed 2018-03-12 by FezVrasta

FezVrasta, 2018-03-12:
@manup would this smart power outlet work as repeated with DeCONZ?

https://www.osram-lamps.com/ecatalog/smart-home/smart-home-components/smart-plug/index.jsp

manup, 2018-03-12:
Jep, it's a popular one too, have one on my own.