dresden-elektronik/deconz-rest-plugin #471: How to create rules with the new daylight sensor?

Krocko, 2018-03-12:
If i create a rule for sunrise-sunset
```
      {
        "address": "/sensors/14/state/status",
        "operator": "gt",
        "value": "139"
      },
      {
        "address": "/sensors/14/state/status",
        "operator": "lt",
        "value": "191"
      }
```
this work.

But if i create a rule for sunset-sunrise
```
      {
        "address": "/sensors/14/state/status",
        "operator": "gt",
        "value": "189"
      },
      {
        "address": "/sensors/14/state/status",
        "operator": "lt",
        "value": "141"
      }

```
this don't work.
How can we solve this?

manup, 2018-03-12:
I've stumpled over the same issue :)
You need to add a trigger:

```json
      {
        "address": "/sensors/14/state/status",
        "operator": "gt",
        "value": "139"
      },
      {
        "address": "/sensors/14/state/status",
        "operator": "lt",
        "value": "191"
      },
      {
        "address": "/sensors/14/state/status",
        "operator": "dx"
      }
```

```json
     {
        "address": "/sensors/14/state/status",
        "operator": "gt",
        "value": "189"
      },
      {
        "address": "/sensors/14/state/status",
        "operator": "lt",
        "value": "141"
      },
      {
        "address": "/sensors/14/state/status",
        "operator": "dx"
      }
```

ebaauw, 2018-03-12:
You need to create to separate rules for logical "or"; the conditions in a single rule are combined by logical "and".  It would make sense to support the "in" operator here, with the same wrapping logic as for `config.localtime`.

manup, 2018-03-12:
But isn't the `in` rule limited to just one range?

"Only one "in" operator is allowed in a rule."

Krocko, 2018-03-12:
>I've stumpled over the same issue :)
You need to add a trigger:

      {
        "address": "/sensors/14/state/status",
        "operator": "gt",
        "value": "139"
      },
      {
        "address": "/sensors/14/state/status",
        "operator": "lt",
        "value": "191"
      },
      {
        "address": "/sensors/14/state/status",
        "operator": "dx"
      }
     {
        "address": "/sensors/14/state/status",
        "operator": "gt",
        "value": "141"
      },
      {
        "address": "/sensors/14/state/status",
        "operator": "lt",
        "value": "191"
      },
      {
        "address": "/sensors/14/state/status",
        "operator": "dx"
      }
Note I've switched the values in the last rule.

This is both sunset-sunrise. This works for me without ```
{
        "address": "/sensors/14/state/status",
        "operator": "dx"
      }
```
But how can i set sunrise-sunset (night)?

manup, 2018-03-12:
Can you provide the complete rule conditions? Should the rule only trigger once night begins or combined with another event like a motion sensor or switch button press?

Krocko, 2018-03-12:
```
{
    "actions": [
      {
        "address": "/groups/5/scenes/2/recall",
        "body": {
        },
        "method": "PUT"
      },
      {
        "address": "/sensors/7/state",
        "body": {
          "status": 2
        },
        "method": "PUT"
      }
    ],
    "conditions": [
      {
        "address": "/sensors/7/state/status",
        "operator": "eq",
        "value": "1"
      },
      {
        "address": "/sensors/7/state/status",
        "operator": "dx"
      }
    ],
    "name": "Bewegung Flur nachts",
    "periodic": 0,
    "status": "enabled"
  }
```

manup, 2018-03-12:
Indeed that would simplify some things a lot shouldn't be too difficult to add. But on the downside this wouldn't be compatible to the Philips API.

ebaauw, 2018-03-12:
> But on the downside this wouldn't be compatible to the Philips API.

Neither is `state.status` for the Daylight sensor...

ebaauw, 2018-05-04:
> You need to add a trigger:

Same for a condition on `state.daylight`.  Rule doesn't fire.  I added a `dx` condition on `state.lastupdated`, but need to wait till tomorrow to see if it works...

Krocko, 2018-05-04:
A rule with a condition `state.daylight` works for me, without problems.

ebaauw, 2018-05-05:
@Krocko does the condition on `state.daylight` actually trigger the rule?

@manup, I think there's a bug: the event is generated for `state.status` instead of for `state.daylight`:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/43f8883ef23b668f1e763d0052954983d1ac0f3f/rest_configuration.cpp#L2710-L2711

Krocko, 2018-05-05:
Yes the rules trigger. 
See here rule 3 and 4. 
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/529#issuecomment-383579790

manup, 2018-05-07:
>@manup, also, event for state.lastupdated is missing. And Daylight sensor's state shouldn't be restored from database, to it will change on starting deCONZ (as it does on the Hue bridge - this is how I detect when the Hue bridge boots). I'll create a PR, but I better run some tests overnight before you merge it.

Many thanks for looking into it, can I merge the PR now, or should I wait a bit?
2.05.26 could arrive around Wednesday.

ebaauw, 2018-05-07:
Tests seem fine, PR can be merged, thanks.

ooii, 2018-05-11:
I assume `sensors/14` mentioned by @Krocko is the `daylight sensor`. What does the `status` value mean? I could not find any doc about how to use that.
Thanks.

ebaauw, 2018-05-11:
It's the last daylight-releated event, see https://github.com/dresden-elektronik/deconz-rest-plugin/blob/51f6c763976872128f6ff9548fa778587b092353/daylight.h#L14-L27

homebridge-hue also exposes this and the current period as string values, but for rules/automations you'd need the numeric value.  I actually find I only use `state.daylight` and `state.dark` (added with my PR) in rules/automations.

ooii, 2018-05-11:
Thanks.
I assume (and read in another discussion) that I need to set `lat` and `long` somewhere but it's not in `config`. Where can I do that?

closed 2019-03-30 by stale[bot]