dresden-elektronik/deconz-rest-plugin #472: False signals from Tradfri remotes?

FezVrasta, 2018-03-12:
Sometimes my lights will turn on by themselves and I suspect it's because the Tradfri remotes are triggering signals without me pressing any button.

My setup is:

- DeCONZ (RPi3 + RaspBee) with 3 remotes paired
- Hue Bridge 2.0 with several bulbs paired
- Home Assistant that makes the two communicate

I'd like to know if there's any logging I can inspect to understand if what I suspect is true?

svenove, 2018-03-12:
I experience the same each time deCONZ is restarted. It seems the last button on the remote is treated as if it was re-pressed. I'm using Homeseer and JowiHue-plugin.

Could this be what you are experiencing?

manup, 2018-03-12:
From deCONZ view it seems to be correct, since just the last known state is reported in the REST-API also after a deCONZ restart.

I'm not sure how HA and JowiHue process the events but if the `state.lastupdated` is taken into account there should be no ghost events.

Kane610, 2018-03-12:
@manup HA doesn't care for that right now, is that recommended to get rid of false positives when restarting deconz?

wvuyk, 2018-03-12:
@manup JowiHue does take lastupdated into account for certain sensors. For several remotes a new lastupdated time is send, even with reports without real updates, so these quite are hard to ignore? If lastupdated would be set to report the real last update it would be a lot easier to ignore updates.

manup, 2018-03-12:
> @manup HA doesn't care for that right now, is that recommended to get rid of false positives when restarting deconz?

It depends how events in HA are triggered, after restart deCONZ won't trigger an button press event but there might be websocket events for `config.reachable` and `config.battery` which should not trigger any button press logic. 

For remotes the `state.lastupdated` is *only* refreshed if a button was pressed (also the same button again) and it should be safe to use. If `state.lastupdated` is changed without a button being pressed when were would be an issue in deCONZ which needs to be fixed.

manup, 2018-03-12:
> @manup JowiHue does take lastupdated into account for certain sensors, but not all.

> For several remotes a new lastupdated time is send, even with reports without real updates, so these are quite hard to ignore?

That shouldn't happen, see above the `state.lastupdated` is (should) happen for `state.buttonevent` only. Do you know which switches have this issue?

> Especially when you need to enable double presses and such. If lastupdated would be set to report the real last update time it would be a lot easier to ignore these.

Hmm I don't understand, this is exactly what they do, if not we need to figure out which switches have this problem or which other attributes may interfere.

Kane610, 2018-03-13:
@FezVrasta you can enable debug with logger component in hass, then you can check the logs when your spontaneous trigger happens and we should be able to pinpoint if there is something in hass that ghost-fires an event

FezVrasta, 2018-03-13:
The thing is that it's difficult to understand who triggered a light toggle, I can't seem to find a log of the automations, which would be very useful because my buttons are linked to automations to control the lights.

closed 2019-03-30 by stale[bot]