dresden-elektronik/deconz-rest-plugin #473: Firmware update from Phoscon fails

jo-me, 2018-03-12:
Hi, 
I'm running deconz 2.05.08 for a ConBee attached to a Raspberry Pi 2. 
In Phoscon, it suggest a firmware update:

[image]

When I click the update button it apparently starts the update:

[image]

After 10-15s (!) it finishes and says that the update is completed.
However, it shows the old firmware version and even a deconz version that I had never installed:

[image]

After a while, the same firmware update is suggested again.

What is going wrong here?
Are firmware updates actually possible through phoscon?

manup, 2018-03-12:
>What is going wrong here?

Is deCONZ running as root? The current implementation expects that it is running as normal user with sudo rights (like default user pi).

>Are firmware updates actually possible through phoscon?

Yes but as noted we test usually with the standard Raspbian distro and deCONZ installation and default user pi. For other variations we need to adapt the code to support these.

ooii, 2018-03-13:
I have the same issue. 
I tried to run the app as root but same behavior as @jo-me.
Note that I'm running version 2.05.11 of 07/03/2018 under Ubuntu and not as the pi user.

jo-me, 2018-03-13:
I'm running deconz on a hassbian (preconfigured debian/raspbian+ home assistant) installation. Installed it directly on the system. So, it was automatically set up as a systemd service.

The deconz.service file indicates that it is indeed running as pi user:
```
[Service]
User=pi
```

So.. based your assumptions it should've worked, right?

I noticed that there is no such "User" entry in deconz-update.service. Might not be relevant, though.

manup, 2018-03-17:
Good catch, internally the firmware update should just start when one device is attached, however this seems to be a bit badly implemented since it only checks for USB devices with 0x6015 which fails in your case.

I'll put it on the todo list

closed 2019-03-30 by stale[bot]