dresden-elektronik/deconz-rest-plugin #474: Unreachable devices not visible in REST api

jonatanolofsson, 2018-03-13:
Updating from 2.05.08 to 2.05.12, my unpowered lights no longer show in the REST API or in the web interface (not just unreachable). Are they even removed?

ebaauw, 2018-03-13:
See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/97

manup, 2018-03-13:
Showing unpowered lights directly after deCONZ restart is on the todo list, there is a solution how to support that, but it will take a few weeks until there is time for that. Main problem is to not show ghost devices which weren't online for months/years, or even aren't part of the network anymore. Hence the database will be extended with a ttl counter or something like that.

jonatanolofsson, 2018-03-13:
@manup Regarding not showing ghosts, is that desired behaviour? How would one even decide what is too old? The system I am developing is intentionally very static (i.e., dynamic discovery is a no, missing devices are an error), where any 'dynamic' behaviour to the backend has to be worked around/detected as an error, so I personally believe it should be up to the user to remove ghost devices manually. Also, it would save you from having to implement the logic and timeout.

jnmarczi, 2018-03-14:
@jonatanolofsson +1

jonatanolofsson, 2018-05-15:
@manup How is progress on this issue? Right now my software is crashing on restart because devices are missing.. =)
I suppose I _could_ fix my own software but... :D :D (j/k, I'm gonna fix that as well ^^ I do think this is a relevant issue though)

jonatanolofsson, 2018-05-22:
> It's not forgotten but also not around the corner, however in the
> background recent deCONZ versions do collect and store more information of
> lights in the database which is a requirement to implement the feature, so
> there is some progress ;)

Why not simply list all devices in the database?

manup, 2018-05-22:
There are setups out there which run since 2013, with many "dead" lights in the database. I guess people would be confused too see lights again which weren't there for years.

In order to support the feature without surprises we need to take care to only show lights which are not from the stoneage. It sounds easier than it is since some refactoring is required to implement it well.

tunip, 2018-06-19:
Hopefully the implementaion takes not to long. If you use deCONZ together with homebridge-hue, I cant restart my raspberry (deCONZ-gateway) until all my bulbs (19) in 8 rooms are powered on with the (good old) wallswtich. As a workaround I seperated the devices, Philips Hue lights connected to a Hue bridge V2 and the sensors connected to the deCONZ gateway. But the target would be only use one bridge/gateway for all devices which could be restarted in case of software-updates and still works after power outages.

ebaauw, 2018-06-19:
Use a lightlist to delay hoembridge-hue initialisation, until all lights have been discovered, see https://github.com/ebaauw/homebridge-hue/issues/246.

closed 2019-03-30 by stale[bot]