dresden-elektronik/deconz-rest-plugin #477: Gateway on channel 0 and cannot change it

ooii, 2018-03-14:
Hi,

My gateway is on channel 0 and I cannot change it to any other channel. I use the webapp advanced settings to select another channel then hit "Apply". Despite the message "OK! Configuration updated", I still have "Your gateway is on channel 0 and has the network Id 0x0000".
Therefore, it cannot find any of my lights.

manup, 2018-03-14:
Can you provide more details on you setup: RaspBee or ConBee, operating system?
Channel 0 and network 0x0000 means not connected at all, so it seems installation is not complete.

ooii, 2018-03-14:
I'm running deCONZ under Ubuntu and I'm using a ConBee USB Gateway.

manup, 2018-03-14:
See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/94

ooii, 2018-03-14:
> Channel 0 and network 0x0000 means not connected at all, so it seems installation is not complete.
It used to work then stopped. 
Any idea what reason can make the installation not complete now?

manup, 2018-03-14:
Hard to tell, do you have other FTDI based USB devices connected?

Perhaps debug logs shed some light, start deCONZ via:

    deCONZ --dbg-info=1 --dbg-error=1 --http-port=80 -platform minimal

(stop the deconz service first)

ooii, 2018-03-14:
I have another one, RFXtrx433 transmitter, which I stopped using during the firmware upgrade.
I'll test later and keep you informed.
Thanks again.

ooii, 2018-03-14:
The problem comes from another instance (investigating how it has started) of deConz which makes the gateway unavailable for the service I was running.
I can close this issue. Thanks for your help.

closed 2018-03-14 by ooii