dresden-elektronik/deconz-rest-plugin #479: Getting inconsistent data from websocket (xy_color)

tleegaard, 2018-03-14:
Websocket:
```
2018-03-13 21:30:44 DEBUG (MainThread) [pydeconz.utils] Sending {'data': '{"on": true, "xy": [0.4448, 0.4066], "bri": 240}'} to http://192.168.x.x:80/api/xxxxxxxxxx/lights/3/state
2018-03-13 21:30:44 DEBUG (MainThread) [pydeconz.utils] HTTP request response: [{'success': {'/lights/3/state/on': True}}, {'success': {'/lights/3/state/bri': 240}}, {'success': {'/lights/3/state/xy': [0.4448, 0.4066]}}]
2018-03-13 21:30:45 DEBUG (MainThread) [pydeconz.websocket] Websocket data: b'\x81E{"e":"changed","id":"3","r":"lights","state":{"bri":240},"t":"event"}\x81E{"e":"changed","id":"3","r":"lights","state":{"x":29036},"t":"event"}\x81E{"e":"changed","id":"3","r":"lights","state":{"y":26542},"t":"event"}'
2018-03-13 21:30:45 DEBUG (MainThread) [pydeconz.deconzdevice] Panthella: update bri with 240
2018-03-13 21:30:45 DEBUG (MainThread) [pydeconz.deconzdevice] Panthella: update x with 29036
2018-03-13 21:30:45 DEBUG (MainThread) [pydeconz.deconzdevice] Panthella: update y with 26542
```

This gives me different xy-values in Home Assistant from time to time:
```
{
  "brightness": 126,
  "xy_color": [
    29035,
    26541
  ],
  "effect_list": [
    "colorloop"
  ],
  "rgb_color": [
    253,
    169,
    0
  ],
  "friendly_name": "Panthella",
  "supported_features": 125
}
```
and
```
{
  "brightness": 126,
  "xy_color": [
    0.444783,
    0.406578
  ],
  "effect_list": [
    "colorloop"
  ],
  "rgb_color": [
    214,
    177,
    108
  ],
  "friendly_name": "Panthella",
  "supported_features": 125
}
```
And sometimes they even get mixed up:
```
{
  "brightness": 126,
  "xy_color": [
    29035,
    0.406578
  ],
  "effect_list": [
    "colorloop"
  ],
  "rgb_color": [
    255,
    0,
    0
  ],
  "friendly_name": "Panthella",
  "supported_features": 125
}
```
See also: https://github.com/Kane610/deconz/issues/7

ebaauw, 2018-03-14:
The websocket events return the raw ZigBee value, between 0 and 0xfeff, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/32#issuecomment-315549173.  The REST API normalises this value to the range 0.0 to 1.0.  It does so wrongly, by the way, by dividing the ZigBee values by 0xfeff instead of by 0xffff.  Also, the light changes still need to be collected into a single web socket event, as is done for sensor changes.

tleegaard, 2018-03-15:
So the different results for xy above is because of errors in the normalization done by the REST API?

ebaauw, 2018-03-15:
No, they’re the result of the websocket notifications not doing any normalisation at all and (apparently) home assistant not taking that into account.

drbytes, 2018-06-05:
I can't say right away without going trough the deconz code for hass but I have a similar problem. For now, when I get a light update from the websocket I look up the values for xy using the rest states for lights. It's annoying unneeded cycles and code clutter.

Kane610, 2018-06-26:
Based on the comments here I hope to have improved the behavior for next hass release. Thanks guys!

tleegaard, 2018-06-27:
Great! Do you think I could close the issue then?

Kane610, 2018-06-27:
Other people have said it works better with the fix, so it is up to you if you want to close it before you test it out ;)

tleegaard, 2018-10-26:
No issues recently.

closed 2018-10-26 by tleegaard