dresden-elektronik/deconz-rest-plugin #480: Support for Aqara Wall switch QBKG04LM

KingTomaHawk, 2018-03-14:
Hi,

I bought two of the Aqara Single key Wall switches (QBKG04LM) and successfully paired them with deconz, but the device name is shown as 0x0000 and there is no enty in the REST Api. 
Am I doing something wrong, because the Aqara Wall Switches are listed as supported with deconz?

Thanks

KingTomaHawk, 2018-03-18:
Can somebody confirm if the wall switch is supported or not? 
If not, I found someone who seems to have successfuly implemented the wall switch to deconz:

https://github.com/zydeco/deconz-rest-plugin/commit/4d79ea0d08fbd1a32022f033b6fb216a73b7a1d8

manup, 2018-03-18:
Can you post a screenshot of the basic cluster after the switch joins, we need to verify the modelid is registered.

Note to add the switch please start sensor search in Phoscon App then reset the switch.

KingTomaHawk, 2018-03-18:
[image]
Sorry for the late reply. It seems like some update broke my AP feature so it took a little bit longer then expected.

manup, 2018-03-18:
Hmm the lumi.ctrl_neutral1 should be supported and visible in the REST-API (added while sensor search is active), can you please double check that it isn't there?

KingTomaHawk, 2018-03-18:
I can not find it on /api/<apikey>/sensors      :(

tomfritz1, 2018-03-18:
I have the same issue - the switch is added in WebApp but not to Rest-API - neither as Sensor nor as light.

KingTomaHawk, 2018-03-19:
I reseted yesterday my deconz and my Pi to test if there was something wrong with my configuration, but it still doesn't list the wall switch in REST Api.

KingTomaHawk, 2018-04-07:
Any updates on this issue?

KingTomaHawk, 2018-04-24:
FYI: I updated to the new "latest-deconz-beta" build, and noticed that the wall switch is now listed as a light! But, I think it also created 3 other "lights" in phoscon without any function.

ebaauw, 2018-04-24:
See #356.  Could you please post the light resources there.  Which one does work?  And check whether any sensor resources where created?  Thanks.

closed 2018-05-21 by KingTomaHawk

lenisko, 2019-04-30:
Hey there!

Can this device get somehow moved to `Switch` section?
It's Xiaomi `QBKG04LM` Aqara Wall Switch / one key version / can act as router.

[image]