dresden-elektronik/deconz-rest-plugin #488: Xiaomi sensors (door/motion) show 0° temperature in Phoscon

jo-me, 2018-03-17:
[image]

manup, 2018-03-17:
After a while (once per hour) all Xiaomi devices do send the temperature to the gateway and it should be visible.

to0b, 2018-03-17:
Does the motion and door/window sensors have a temperature sensor onboard?

manup, 2018-03-18:
Have a look here, the ones marked with `T` also have a temperature sensor:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices

Be aware they only update temperature once per hour.

zladukas, 2018-03-18:
@manup Ikea Tradfri motion sensor battery is supportes, but there is no B near it.

jo-me, 2018-03-18:
Cool, I did not know that the Aquara sensors had a temperature sensor in them..
This morning, the aquara door sensor had a reasonable value instead of the 0° from yesterday right after the deconz update.

But for the Xiaomi Mi sensors it is still a bug, though. According to your list, they don't have temperature sensor but are still displayed in Phoscon as if they had one. (the sensors in the screenshot above are all Xiaomi Mi ones).

tleegaard, 2018-03-18:
My Aqara door sensor reports 27°C while the Hue motion sensor in the same room reports 22 °C - so I wouldn't trust the readings :-)

ebaauw, 2018-03-18:
I wouldn't trust either of them.  The temperature sensor in the Xiaomi devices (other than the Aqara weather and Mi temperature/humidity) is rather crude.  I find that multiple Hue motion sensors, when placed next to each other, cannot agree on the current temperature either.  "With one watch a man knows what time it is, with two he's never sure..."

I find the Aqara weather sensors are fairly accurate, but sometimes take a long time to send a new attribute report, causing the temperature in deCONZ to lag behind.  The Heiman temperature/humidity sensor seems to do the best job, at least for temperature.

I introduced `config.offset` mainly to calibrate my Hue motion sensors, but the deviations don't seem to be constant over different temperatures.  When I set the offset, so the motion sensor shows the correct temperature at 22˚C, it will still show an incorrect temperature at 17˚C.  Maybe there's a reason Philips doesn't advertise that the motion sensor includes a temperature sensor...

Note that I don't have any professionally calibrated thermometer to compare the sensors to.  I compare against the old-fashed room thermostat of my central heating and a consumer-grade thermo/hygrometer, that came with my wooden floors.

ACrazyConcept, 2018-05-10:
Hello
I got a Xiaomi motion sensor yesterday. It still shows 0˚C.
Is there a way I can verify if it has been connected properly?
In deCONZ it only show two clusters, Basic and Occupancy. No temp attributes in either.

manup, 2018-05-12:
The temperature for the motion sensor is not transferred via temperature cluster but a Xiaomi specific command not visible in the deCONZ gui. The sensor reports the command once per hour.

Is it the Aqara version — which model id has the sensor?

ACrazyConcept, 2018-05-13:
Oh ok.
Actually it is a Mijia. I was not aware there were two version when I ordered. It says RTCGQ01LM and CMIT ID: 2015DP0292

ACrazyConcept, 2018-05-18:
Ok I got myself an Aqara version now as well. It reports a temperature.
Is there a way to get that temperature into Home Assistant when it does not have a cluster in deCONZ?

ebaauw, 2018-05-19:
You’ll have to take that up with the developers of the Home Assistant plugin.  The internal temperature is exposed as `config.temperature`, contrary to the `state.temperature` for the “real” temperature clusters.

ACrazyConcept, 2018-05-19:
Of course. Thanks for the feedback :)

sealjo, 2018-12-28:
> Of course. Thanks for the feedback :)
> Edit: I figured it out. Thanks for pointing me in the right direction.

How did you do it? Have the same issue, I want the temperature I see in phoscon to be shown in HA but it isn't imported by the deconz config.

closed 2019-05-04 by stale[bot]

mcka-dev, 2020-02-13:
My Aqara door sensor reports 34°C  (((
[image]

soundylinz, 2020-08-24:
My sensor reports 35°C without changes for a few hours:

[image]