dresden-elektronik/deconz-rest-plugin #489: [Help needed] Cannot move Hue lights from Hue bridge to deConz

ooii, 2018-03-17:
Hi,

I have a Hue bridge with lights, sensors, and switches. I'm willing to move everything to my deConz setup, based on ConBee stick.
I'm trying to add my Hue bulbs to deConz and I'm facing issues. I have three different behaviors with three bulbs.

- Bulb 1: Never paired with Hue bridge. I can add it with no issue to deConz. It is placed 1 meter far from the ConBee gateway.
- Bulb 2: Already paired with Hue bridge. Placed 2 meters far from the ConBee gateway. When I try to reset it with Hue Dimmer (placing the Dimmer very close to the bulb and pressing the On and Off buttons for 10 seconds), it never blinks but Bulb 1 blinks.
- Bulb 3: Placed around 4 meters from the gateway. When I reset it, it blinks. But when I try to pair it with the gateway, I cannot.

Any idea why I have this different behaviors? And how I can deal with my bulbs to add them? Is it mandatory to bring them one by one very close to the gateway to be able to add them?

Thanks for your help.

ebaauw, 2018-03-17:
Did you power-cycle bulb 2?  With the latest firmware, the Hue bulbs only react to touchlink for a limited time after power on.  For 3, make sure to hold the On and Off buttons until the light stops blinking, or it won’t reset.  You might want to power-off nearby lights, so it’s forced to connect directly to the gateway.

ooii, 2018-03-18:
Thanks Eric, that did it.

> You might want to power-off nearby lights, so it’s forced to connect directly to the gateway.

Does that mean that I need to bring all my bulbs close to the gateway to add them, contrarily to Hue bridge which allows adding them through the mesh?

ebaauw, 2018-03-18:
In a Hue-only environment adding devices throught the mesh works fine, but different manufacturers seem to handle the meshing somewhat differently.  Note that in this context “close” means within 10m or so.  Only for touchlink, you need to be within 15cm or so.

Since most of my lights are not easily moved, I used to power my Raspberry Pi with RaspBee by a power bank and bring it next to the light.  You might also use a USB extention cable to bring the ConBee close to the light.  For sniffing, I connect my ConBee over a 5m female USB-A to USB-A cable to place it between the device and the RaspBee.

ooii, 2018-03-18:
> In a Hue-only environment adding devices throught the mesh works fine

I have only Hue bulbs, so I'm going to add them through the mesh. I'm running my system on a difficult-to-move ubuntu server placed in my house basement while I have lamps in the attic.

ooii, 2018-03-18:
I close it for now.

closed 2018-03-18 by ooii

ooii, 2018-05-02:
> In a Hue-only environment adding devices throught the mesh works fine, but different manufacturers seem to handle the meshing somewhat differently.

I have moved all my Hue environment (45 bulbs, 15 switches, and 8 sensors) to deconz. Now, I'd like to add some Osram plugs. Is there any risk concerning the mesh network if I add them now? Will I be able to add new devices without placing them close to the gateway?