dresden-elektronik/deconz-rest-plugin #490: Direct link for phoscon?

jo-me, 2018-03-18:
Hi

 is there a direct link to the phoscon app of a specific gateway?

I only have one instance, but on first access via the gateway selection / login page is shown with the one gateway that I have.

manup, 2018-03-18:
Yes every gateway have a copy of the app you can access via.

`http://<gw-ip-address>/pwa`

jo-me, 2018-03-18:
Okay, but I'm also getting redirected to 
http://<ip>:<port>/pwa/login.html

So still one more click until the actual gateway page. Btw, I don't have a password set up.

manup, 2018-03-18:
Yep login is required even in local network, if the default password is still set the login page will ask to specify a new password with at least 8 characters length.

closed 2019-03-30 by stale[bot]