dresden-elektronik/deconz-rest-plugin #491: All sensors with ID higher as the Daylight sensor not visible in the Phoscon app

Krocko, 2018-03-18:
The new Daylight sensor has the ID 14 for me. All sensor with ID higher then 14 are no longer visible in the Phoscon app.

ebaauw, 2018-03-18:
I put the Daylight sensor on `/sensors/1` (just as the Hue bridge) and all my other sensors and switches show in Phoscon.  deCONZ v2.5.13.

Krocko, 2018-03-18:
Strange. I am on deCONZ v2.5.14. 
First all sensors there visible in Phoscon, but after some seconds all sensors with ID higher 14 are gone.

manup, 2018-03-18:
Indeed strange, I recon they are visible in REST-API?

Might be an Javascript exception thrown in Phoscon App during build of the sensors list.
Which browser do you using — if possible you can check the browser Javascript console if there is an exception error displayed?

Krocko, 2018-03-18:
Yes, the sensors are all in the REST-API and in the GUI.
I use only the Safari browser on all my devices.
[image]

manup, 2018-03-18:
Found the bug, I've updated the packages and http://dresden-elektronik.de/pwa may take a bit until browser does refresh the cache.

Krocko, 2018-03-18:
Thank you Manuel. Problem solved.

closed 2018-03-18 by Krocko